libplctag/libplctag #358: AB Logix Read Extended Tag Properties

PWSys, 2022-03-26:
Newer firmware AB Logix PLCs provide for extended tag properties.  This includes engineering units, and min and max values.  Can these properties be read by the library?  Or would it be possible to read tag descriptions?

Using Rockwell drivers, they can be referenced using the following syntax tagname.@property.  The following is an example:
Tag Name: V_01
Tag Description: Tank Outlet Valve
The tag description can be read by using the following address: "V_01.@Description".

kyle-github, 2022-03-26:
The latest firmware I have access to is v31.   Does that support this?

If it is done via the tag name, I can make it work.   That is how the alarm stuff works.   

Let me think about how to do this.   Would you be able to test?

PWSys, 2022-03-26:
Thanks for looking into this!

It is supported in version 31.  It looks like it was introduced in 21.

These are the available properties:
- .@Description: Description of the tag
- .@EngineeringUnit: A system of measurements of physical quantities
- .@Min: The minimum numeric value for the non-Boolean tag being measured
- .@Max: The maximum numeric value for the non-Boolean tag being measured
- .@State0: The false Boolean state of the tag
- .@State1: The true Boolean state of the tag

There are additional properties available in 33.
- .@Library: Associates instructions with a display library for faceplate call-up.
- .@Instruction: Associates the instruction name for faceplate call-up.
- .@Label: A label that appears on the graphic HMI symbol.
- .@Area: The area name for area-based security in the HMI.
- .@URL: The Uniform Resource Locator (URL) for accessing the help content from HMI.
- .@Navigation: A path to another source of information. For example, a document link or website.
- .@Name: A text string ranging from 0 through 512 characters that defines the tag name in client application.
- .@Quality: A string following the OPC DA standard to indicate the tag status.

I can also test.

kyle-github, 2022-11-13:
Sorry for dropping this one for so long.   Have you tried to see if this works?   I see in the currently released code that the "@" character is accepted in symbolic names.

PWSys, 2022-11-27:
@kyle-github , I did give this a try without success.  The PLC tag status is changing from a 1 to -7 (PLCTAG_ERR_BAD_PARAM).

I created a PLC controller named Test_DINT of type DINT with a Max attribute of 100.  With the following handle I am able to successfully read the current value of this tag.
`"protocol=ab_eip&gateway=192.168.1.101&path=1,0&plc=LGX&elem_count=1&name=Test_DINT"`

If I append ".@Max" to the handle, "-7" is returned.

`"protocol=ab_eip&gateway=192.168.1.101&path=1,0&plc=LGX&elem_count=1&name=Test_DINT.@Max"`

Changing the case of max or adding elem_size=4 doesn't have an impact.

The address `Test_DINT.@Max` does work in FactoryTalk Linx Enterprise.

Is there anything else I should try?

In case it is helpful an excerpt from the manual is below and it indicates that the Max value should be a DINT.

[image]

MikelAyani, 2025-04-12:
Did anyone get the properties somehow? I am interested in reading them too.

kyle-github, 2025-04-12:
Marking this as a bug.   Should work.

MikelAyani, 2025-04-14:
I did some experiments these days but I cannot get it working with a Compatlogix 5380 v37.
I can read and write in the tag test_DINT, both as DINT and independent bits. But trying any property returns an error.
I am mainly interested in gathering the description, both at DINT and bit level.

kyle-github, 2025-04-24:
@MikelAyani, Can you enable debug level 4 and capture the logs?   Then I can see if the error is coming from the PLC or from within libplctag.   Thanks.

MikelAyani, 2025-04-24:
Sure, Do you want me to paste the code and the result here or you want me to send it to you? I already have it.

MikelAyani, 2025-04-24:
**The code:**
`from plctag import libplctag

libplctag.plcTagSetDebugLevel(4)
path = "protocol=ab_eip&gateway=192.168.0.53&path=1,0&cpu=LGX&elem_size=1&name=test_DINT.@Description"
tag = libplctag.plc_tag_create(path, 2000)
print(tag)
rc = libplctag.plc_tag_status(tag)
print("Status:",rc)
tag_size = libplctag.plc_tag_get_size(tag)
print("Size:", tag_size)
rc = libplctag.plc_tag_read(tag, 2000)
print("Read:", rc)`

**The error output:**
...
2025-04-24 20:39:34.031 thread(2) tag(0) WARN check_read_status_connected:1311 CIP read failed with status: 0x5 Path Destination Error
2025-04-24 20:39:34.031 thread(2) tag(0) INFO check_read_status_connected:1312 The Class, Instance or Attribute value specified in the Unconnected Explicit Message request is incorrect or not supported in the Target device. Check the manufacturer's documentation for the correct codes to use.

MikelAyani, 2025-04-24:
There you have:
[logs.txt](https://github.com/user-attachments/files/19897820/logs.txt)

MikelAyani, 2025-04-24:
In case it helps, I have enabled the Min and Max extended properties in the tag but trying test_DINT.@Min or Max gives the same error.

[image]

[logs.txt](https://github.com/user-attachments/files/19897864/logs.txt)

kyle-github, 2025-04-25:
Hmm, this error is coming from the PLC itself.   It does not like encoding the "@Max" part as a symbolic segment like the tag name.   So I think that there must be something else that is going on here.   If you can use Wireshark to capture some of this, it could be very helpful. I have a suspicion that the Rockwell tooling is doing some sort of translation here.  It might be translating into attribute accesses?  It might be translating the name in another way.

The way that the UI shows these as "extended properties" is suggestive that Rockwell is treating them differently than the alarms which work with the "@" syntax.