libplctag/libplctag #360: Read and Write Wall Clock Time on ControlLogix

airwedge1, 2022-05-11:
Anyone know of a way to read and write the wall clock time on a ControlLogix PLC.  I know the @tags special tag exists. Maybe there is something similar for the wall clock time?

airwedge1, 2022-05-19:
Thanks for your response Kyle.  In ladder logic you use a special GSV command to get the wall clock time.  I know other libraries are able to read and write to the clock so it is possible.  This library does: https://www.advancedhmi.com/index.php?main_page=product_info&cPath=3&products_id=14 (but it has other problems) I'll try and do some more research to try and figure out how it's done over EIP.  I know the documentation can be limited at times with these sorts of things.

airwedge1, 2022-08-16:
I came across a different library that implements this pycomm3. I bet the below is enough documentation to figure it out. 
 If you pointed me in the write direction on how you would implement this I can probably contribute. 

https://docs.pycomm3.dev/en/latest/_modules/pycomm3/logix_driver.html#LogixDriver.get_plc_time

wall_clock_time = b"\x8b"  # Micro800 CIP client messaging quick start

def get_plc_time(self, fmt: str = "%A, %B %d, %Y %I:%M:%S%p") -> Tag:
        """
        Gets the current time of the PLC system clock. The ``value`` attribute will be a dict containing the time in
        3 different forms, *datetime* is a Python datetime.datetime object, *microseconds* is the integer value epoch time,
        and *string* is the *datetime* formatted using ``strftime`` and the ``fmt`` parameter.

        :param fmt: format string for converting the time to a string
        :return: a Tag object with the current time
        """
        tag = self.generic_message(
            service=Services.get_attribute_list,
            class_code=ClassCode.wall_clock_time,
            instance=b"\x01",
            request_data=b"\x01\x00\x0B\x00",
            data_type=Struct(n_bytes(6), ULINT("µs")),
        )
        if tag:
            _time = datetime.datetime(1970, 1, 1) + datetime.timedelta(microseconds=tag.value["µs"])
            value = {
                "datetime": _time,
                "microseconds": tag.value["µs"],
                "string": _time.strftime(fmt),
            }
        else:
            value = None
        return Tag("get_plc_time", value, None, error=tag.error)

  def set_plc_time(self, microseconds: Optional[int] = None) -> Tag:
        """
        Set the time of the PLC system clock.

        :param microseconds: None to use client PC clock, else timestamp in microseconds to set the PLC clock to
        :return: Tag with status of request
        """
        if microseconds is None:
            microseconds = int(time.time() * SEC_TO_US)

        _struct = Struct(UINT, UINT, ULINT)

        return self.generic_message(
            service=Services.set_attribute_list,
            class_code=ClassCode.wall_clock_time,
            instance=b"\x01",
            request_data=_struct.encode(
                [1, 6, microseconds]
            ),  # attribute count 1, attribute #6, time
            name="set_plc_time",
        )

kyle-github, 2022-08-17:
Interesting.   I did not realize that there was a special CIP command for that.

There's no need to write extra code, just use a raw CIP tag.  I might give that a try in the next couple of weeks.   It would be an example program that used raw tags.  I already have a much more complicated example that does part of tag listing.

dqanxy, 2023-05-09:
Hi, wanted to share the code I made to get/set the wall time. You can use the @raw tag and use the payload
`var rawPayload = new byte[] {
    0x03,
    0x03,
    0x20, 0x8b,
    0x25, 0x00, 0x01, 0x00,
    0x01, 0x00,
    0x0B, 0x00,
};`
for reading and
`var rawPayload = new byte[] {
                0x04,
                0x03,
                0x20, 0x8b,
                0x25, 0x00, 0x01, 0x00,
                0x01, 0x00,
                0x0B, 0x00,
            };`

appended by 8 bytes indicating the wall time in microseconds. These are good references for format: https://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm020_-en-p.pdf
            https://docs.pycomm3.dev/en/latest/cip_reference.html
            https://docs.pycomm3.dev/en/latest/_modules/pycomm3/logix_driver.html#LogixDriver.get_plc_time
            
 `async protected Task<byte[]> SendRawPayloadAsync(byte[] payload)
        {
            //We create @raw to indicate a raw payload rather than a tag.
            var rawTag = new Tag()
            {
                Gateway = "10.20.30.40",
                Path = "1,0",
                PlcType = PlcType.ControlLogix,
                Protocol = Protocol.ab_eip,
                Timeout = TimeSpan.FromMilliseconds(1000),
                Name = "@raw"
            };
            await rawTag.InitializeAsync();
            rawTag.SetSize(payload.Length);
            for (int ii = 0; ii < payload.Length; ii++)
            {
                rawTag.SetUInt8(ii, payload[ii]);
            }
            try
            {
                //Sends the payload to the PLC.
                await rawTag.WriteAsync();
                return rawTag.GetBuffer();
            }
            catch (Exception ex)
            {
                Console.WriteLine("Failed to send packet.");
                Console.WriteLine(ex.ToString());
                return new byte[0];
            }
        }

        /// <summary>
        /// Reads the current wall time of the PLC as a DateTime.
        /// </summary>
        /// <returns>The wall time of the PLC.</returns>
        public async Task<DateTime> ReadWallTime()
        {
            /*https://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm020_-en-p.pdf
            https://docs.pycomm3.dev/en/latest/cip_reference.html
            https://docs.pycomm3.dev/en/latest/_modules/pycomm3/logix_driver.html#LogixDriver.get_plc_time
            See the first document for details on payloads, and the second document for a list of services/class codes/etc.. The third document is a python script with the correct class code/instance/etc.
            The first byte is the service code (0x03 = get attribute list), second byte is the length of the path.
            Path contains the class code and instance number. We encode 0x8b20 = class code, and 0x00010025 = instance 1. (see logical segments in page 14 of the first document for details)
            Next is the number of attributes, we encode 0x0001 = 1 attribute. We then encode the specific attribute, 0x000B being the the wall clock time. */
            var rawPayload = new byte[] {
                0x03,
[... 25 lines cut ...]
            https://docs.pycomm3.dev/en/latest/_modules/pycomm3/logix_driver.html#LogixDriver.get_plc_time
            See the first document for details on payloads, and the second document for a list of services/class codes/etc.. The third document is a python script with the correct class code/instance/etc.
            The first byte is the service code (0x04 = set attribute list), second byte is the length of the path.
            Path contains the class code and instance number. We encode 0x8b20 = class code, and 0x00010025 = instance 1. (see logical segments in page 14 of the first document for details)
            Next is the number of attributes, we encode 0x0001 = 1 attribute. We then encode the specific attribute, 0x000B being the the wall clock time. */
            var rawPayload = new byte[] {
                0x04,
                0x03,
                0x20, 0x8b,
                0x25, 0x00, 0x01, 0x00,
                0x01, 0x00,
                0x0B, 0x00,
            };
            //We concatenate the time in microseconds to our payload.
            var bytes = BitConverter.GetBytes(ConvertDateTimeWrite(time));
            rawPayload = rawPayload.Concat<byte>(bytes).ToArray<byte>();
            var tagResult = await SendRawPayloadAsync(rawPayload);
            if (tagResult.Length == 0) return;
            //Second to last byte will be 0x0C instead of 0x00 if failed.
            if (tagResult[tagResult.Length - 2] != 0)
            {
                Console.WriteLine("Write Wall Time on cooldown. Please wait around 15 seconds before setting the wall time again!");
            }
            Console.WriteLine("Set wall time to " + time.ToString());
        }`

joed74, 2023-07-05:
Doesn't know what `ConvertForRead` do, but to get the correct time out of the response i use:

```
// 1.1.1970 is 621355968000000000 in ticks, 1µs = 10 ticks
var us_since_epoch = BitConverter.ToInt64(response, response.Length - 8);                       
return new DateTime(621355968000000000+us_since_epoch*10);
```

dqanxy, 2023-07-05:
Yeah that's pretty much what ConvertForRead did, sorry for not mentioning it earlier. Logix stores their times in microseconds from 1/1/1970.