libplctag/libplctag #363: Question: usage with golang wrapper

TheFern2, 2022-06-20:
I've added the go wrapper go file to C:\Program Files\Go\src\plctag, then I was getting a pkg-config error, then I installed pkg-config lite with choco, finally added the pc file to pkg-config path. I realize plctag isn't found, but unsure how to fix it. Or if I am missing a step. Any help is greatly appreciated.

I am on win10 x64

```
D:\git\libplctag\src\wrappers\go>go run examples\toogle_bool.go
# plctag
C:/TDM-GCC-64/bin/../lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lplctag
collect2.exe: error: ld returned 1 exit status
```

TheFern2, 2022-06-21:
@alimon @garciaolais could any of you refer to me what the setup looks like on windows 10 or any golang documentation on how to setup c libs on windows 10 for golang usage, if you have same OS?

kyle-github, 2022-06-22:
I see MinGW in the path there.  Is Go on Windows based on that?

The default build of libplctag uses Visual Studio on Windows.   Can a DLL created with Visual Studio work with the loader from MinGW?

There are some half-baked entries in the CMake file to build libplctag with MinGW but it forces 32-bit builds.

If it is a problem of finding the DLL at all, you can try placing it in the usual locations where Windows will look by default.   Perhaps [this](https://docs.microsoft.com/en-us/windows/win32/dlls/dynamic-link-library-search-order) can help?

I wonder if something recently changed in Win10 is breaking the search because the safe search flag is on?

TheFern2, 2022-06-22:
I've given all hope for using this wrapper on windows is too much pain. I was able to run it within minutes on linux. I am now trying to cross compile the example in linux with go build to windows 64, but I am having issues compiling libplctag with x86_64-w64-mingw32-gcc. Maybe what I am doing is not possible.

Trying to build with mingw32 in linux:
```
CC=x86_64-w64-mingw32-gcc cmake .. -DCMAKE_BUILD_TYPE=Release
[snip]
[ 12%] Building C object CMakeFiles/plctag_dyn.dir/src/platform/posix/platform.c.o
/home/fernandob/git/libplctag/src/platform/posix/platform.c:50:10: fatal error: sys/socket.h: No such file or directory
   50 | #include <sys/socket.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/plctag_dyn.dir/build.make:375: CMakeFiles/plctag_dyn.dir/src/platform/posix/platform.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:642: CMakeFiles/plctag_dyn.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```

Trying to crosscompile to win64, I think it won't work until I compile libplctag with mingw:
```
GOOS=windows GOARCH=amd64 CGO_ENABLED=1 CXX=x86_64-w64-mingw32-g++ CC=x86_64-w64-mingw32-gcc go build examples/toogle_bool.go

/usr/bin/x86_64-w64-mingw32-ld: $WORK/b041/_x002.o: in function `_cgo_6a77fbde925a_Cfunc_plc_tag_abort':
/tmp/go-build/cgo-gcc-prolog:66: undefined reference to `__imp_plc_tag_abort'
/usr/bin/x86_64-w64-mingw32-ld: $WORK/b041/_x002.o: in function `_cgo_6a77fbde925a_Cfunc_plc_tag_create':
/tmp/go-build/cgo-gcc-prolog:87: undefined reference to `__imp_plc_tag_create'
/usr/bin/x86_64-w64-mingw32-ld: $WORK/b041/_x002.o: in function `_cgo_6a77fbde925a_Cfunc_plc_tag_decode_error':
```

kyle-github, 2022-06-24:
I haven't had a chance to check this out more.   Is it possible to build the DLL on Windows using Visual Studio and then use that DLL from within Go?   I have never tried to use Go on Windows but I will start learning.

TheFern2, 2022-06-24:
I have built libplctag both on vs2019 and mingw, everything builds no problem. Problem is when trying to run the wrapper example I have no idea where to put the dll, error seen above doesn't tell me anything other than library isn't found. I haven't found any decent tutorial online on what to do for this case.

On linux I just did, cmake, make, make install with gcc compiler. I haven't figured out how to compile libplctag with mingw on linux. From what I read is far easier to work with go and cgo cross compile from linux, than windows. cgo is what is used when using c libraries with the go C package which is used in the wrapper, if you don't use C package then cross compile is pretty easy on any OS to any OS binary.

TheFern2, 2022-07-02:
Okay maybe linux cross compile is more of a pain with cgo. If anyone has any idea how to use the wrapper on windows I'm all ears. I've ran out of ideas on my end. I've searched and read most cgo windows tutorials which aren't many on the web.

kyle-github, 2022-07-02:
I tried to find some notes on this online.   Wow.  That looks painful.   The docs I am finding are all fairly old, 2018 was about the latest one.   

I dug into this a bit more and I think I have identified some issues that I might be able work around.  Using GCC on Windows causes CMake to either pick MinGW (which will compile 32-bit in the default CMakeFiles config file) or think it is really a POSIX OS.   My logic in the CMake config is not great as I pretty much use compiler and platform interchangeably.  This means that it is trying to use the wrong platform.c file, the POSIX one, which is why you got those compile issues.

Part of the problem here is that the compiler selection in the main CMake config is different from how the selection of the platform is done.   I never really noticed that I managed to conflate this so badly before.   I really need to clean that all up a lot as the handling is mashed all together.  Cross compilation is really not handled very well.  I always just used the "native" compiler chain on the platform.

I can't believe that all the docs I find online tell you to use MinGW for cgo.   That's nuts.   No support for Visual Studio?   Even now?   Microsoft did a terrible job of supporting C99 until a few years ago, but Visual Studio does pretty well now including native support for CMake.   It still doesn't support the latest C standards, but it is not as terrible as it was.

I will see if I can get something working on my Windows laptop.   It has Windows 11, but hopefully that is close enough to get something working.

TheFern2, 2022-07-02:
Yeah I agree documentation is very scarce when it comes to golang and cgo. For the most part cross compilation is relatively easy until you bring up cgo to the picture.

I guess not enough people use it to document it well or there are too many assumptions about what people know about c compiling.

Either way I'd expect it to be well documented from Google since is a nice way to bring in wrappers into go from established c drivers.

TheFern2, 2022-07-02:
Running the wrapper was a breeze with linux (Ubuntu 20.04). The only error that got on my way of windows cross compile was the previous post which could be a cmake if statement issue as you mentioned. Hopefully is an easy fix on the cmake file.

I tend to think stuff is easier to do on linux and I had found several reddit posts also suggesting that. But either way would be nice to have a good guide for windows.

https://www.reddit.com/r/golang/comments/p7u6ui/comment/h9mkk2s/

kyle-github, 2022-07-02:
I definitely find development easier on Linux or macOS (though Apple tries hard to make things annoying).  But Windows is definitely a primary platform for a lot of people so I find it strange that Go seems to still treat it as a second class OS.

I've got my Windows laptop charging.  I haven't used it since school ended and my robotics team stopped meeting for the summer.  I'll play with that to see what I need to do.

I wonder if the core problem is just one of naming?   Windows usually looks for libraries without the leading "lib" but maybe the MinGW linker emulates what most Unix-like systems do?  Or maybe multiple files?   Windows uses plctag.dll and plctag.lib when you are trying to link.   Perhaps both are needed?

TheFern2, 2022-07-02:
I am going over my [notes for sfml golang](https://gist.github.com/TheFern2/65e185cbc6f2399fd14001dc94431209) it uses c bindings so maybe I can use the same steps. Maybe I messed up the ldflags. I'll keep you posted.

kyle-github, 2022-07-02:
I found this link: https://stackoverflow.com/questions/50347947/cgo-on-windows-naming-of-loaded-dll

This seems to indicate that there is a special set of settings that you need to do to let cgo's linker find the libraries.

TheFern2, 2022-07-02:
I am not on my win10 machine to show you my build.bat, but I did added ldflags and cflags to a build.bat ldflags points to the libplctag/lib dir and cflags to the header file. I tried both with build.bat and environment flags just like I did for csfml but no luck with plctag.

kyle-github, 2022-07-02:
💩

kyle-github, 2022-07-02:
I am not making much progress getting my Windows laptop running with Go.   I am still installing OS patches.

TheFern2, 2022-07-03:
Okay finally I got it to run with mingw64. I'll make some detailed notes in the next few days. The big step I was missing was manually copying lib, dll, header file to mingw64 lib/bin/include folders which could probably be automated after compilation. I used files from 2.5.1 release.

I'll test tomorrow with my plc.
```
PS C:\Users\fernandob\GolandProjects\testPlcTag> go run .\main.go
ERROR PLCTAG_ERR_TIMEOUT: Could not create tag!
PS C:\Users\fernandob\GolandProjects\testPlcTag> .\main.exe
ERROR PLCTAG_ERR_TIMEOUT: Could not create tag!
PS C:\Users\fernandob\GolandProjects\testPlcTag>
```

One thing to note is that if you use pkg-config lite on windows the libplctag.pc file uses "Program Files" the space doesn't work when trying to build the go program. So I had to add quotes to the path, or for now I just used another path without spaces.

kyle-github, 2022-07-03:
Wow!   That's great!

Is this something that could be added to the new Go wrapper repo or are there changes I can make to the core lib CMake config?   Both?   Generating the libplctag.pc file is done in the original CMake config, so I think that change will need to be there.   Weird that it does not automatically quote the path.

kyle-github, 2022-07-03:
If you don't have a PLC to test with, there is a test program that is built called ab_server that will let you simulate simple tags and arrays of atomic data (DINT, REAL etc.).

TheFern2, 2022-07-03:
I have two Rockwell VMs a softlogix and a emulator VM but for whatever reason they been crashing on my home win10 machine, but works fine on my work laptop win10 and home linux, but is good to know about the ab_server.

TheFern2, 2022-07-03:
> Wow! That's great!
>
> Is this something that could be added to the new Go wrapper repo or are there changes I can make to the core lib CMake config? Both? Generating the libplctag.pc file is done in [...]

Yeah this need to be handled within the core lib install preferably, could also add some scripts to copy dll/header/include to mingw, or whatever compiler all you need is the root path of the compiler where bin/lib/include dirs are and then it should be easy to (transfer/install) libplctag.

kyle-github, 2022-07-03:
For the Java wrapper, I have the Gradle build system go out and download the core library from GitHub, unpack it and then use it.

TheFern2, 2022-07-08:
Closing please read [goplctag docs](https://github.com/libplctag/goplctag/tree/main/docs)

closed 2022-07-08 by TheFern2