libplctag/libplctag #364: Rockwell PLC5 DataHightway+ node 0

jimk907, 2022-06-29:
I very much appreciate this library and have been using it for over a year. Recently I encountered an issue connecting to a PLC5 on Datahighway+ node 0. Even though the connection path routes over Ethernet to a gateway, across a remote I/O adapter to a group of PLC5s, everything works well for PLCs on every other node except the PLC5 on node 0. Of course, RSLinx, and RSLogix5 have no problem connecting to it through the gateway. I suspect an issue in LibPLCTag, but don't have time right now to debug the library code. Here is the gateway and path string portion of the connection string I am using that works for all other nodes:
gateway=xx.x.xx.xxx&path=1,3,A:27:2  <- this 2 is node 2 which works well.
gateway=xx.x.xx.xxx&path=1,3,A:27:0  <- when set to node 0 it returns a timeout error.

kyle-github, 2022-06-29:
Can you run a test with debug level set to 4 and capture the output?   It would be great if you can get the output for both the working case with node 2 and with node 0.

What platform are you on?

My PLC/5 is on node 1.  Perhaps I can set it to zero.   For some reason I seem to remember that zero is special but perhaps I am conflating that with something else.

kyle-github, 2022-07-01:
You can just use the tag_rw2 tool that comes with the core C library.   It is one of the examples.   You can use it to run tests.

```text
tag_rw2 --type=uint8 '--tag=protocol=ab_eip&gateway=10.206.1.40&path=1,2,A:27:1&cpu=plc5&elem_count=1&elem_size=2&name=B3:0/10' --debug=4
```

This bridges across my DHRIO module in my ControlLogix chassis onto DH+ to my PLC/5 at node ID 1.

jimk907, 2022-07-07:
The attached zip file shows the results of two trials as you suggested above. This was done by redirecting stdout and stderr to a file, first using node 2 for the PLC. This trial succeeded. The second trial used node 0 for the PLC. This trial failed with -32.

[libplctag_node0.zip](https://github.com/libplctag/libplctag/files/9063549/libplctag_node0.zip)

jimk907, 2022-07-07:
This picture shows the System Comms dialog from RSLogix5 when connected to the PLC on node 0.
[image]

kyle-github, 2022-07-09:
This might take a bit.   It looks like there are a couple things going wrong.  I _think_ it is in the library, but I am still not sure.   I see the PLC returning an invalid connection size error which should not be happening.

There seems to be a warning earlier as well, so maybe the zero is causing a problem in some of the parsing?

These are very helpful error logs and I should be able to get some traction with them.

kyle-github, 2022-07-11:
Going to take a bit more.   I am traveling for about two weeks.  Sorry for the delay.

kyle-github, 2022-11-13:
I totally dropped this on the floor.   My fault :-(

I looked through the logs and the interesting thing here is that the connection string with the PLC at node 0 is trying to do a Forward Open (CIP connection set up) with a different size than the one with the node ID of 2.   That is definitely wrong, but I cannot see why this would happen.

This is definitely a bug, but I cannot tell where yet.

kyle-github, 2022-11-13:
When I use my PLC as a target (node 1) I see the same connection size as the working example.  So there is definitely a problem here.

kyle-github, 2022-11-13:
What a difference a few months makes from a problem.   This was due to using the destination node ID as a flag on which to determine if a connection was bridging to DH+ or not.   

At least with my PLC/5, this works when I use a node ID of 0 (the PLC doesn't have that, but the path does).   Interestingly, the error I get back is really misleading when the node ID that is sent is not the node ID of the PLC.

Can you test the `prerelease` branch?  I aim to fix a couple other things before I make another release.

kyle-github, 2022-11-14:
This should be fixed in release [v2.5.4](https://github.com/libplctag/libplctag/releases/tag/v2.5.4).

closed 2022-11-14 by kyle-github