libplctag/libplctag #373: Proposing to have the golang wrapper on its own repo for easier download

TheFern2, 2022-07-02:
The current wrapper for go isn't easy to install as a project dependency with `go get` since it isn't really setup as a lib is missing a go.mod file and a url to point to its source. It could be setup on its current dir possibly, I am not entirely sure, but if you look here I ran a quick test  https://github.com/TheFern2/goplctag, I just added a go.mod file, url in go.mod

The current wrapper you have to copy into GOROOT, then reinitialize with `sudo ldconfig` and that sets it up as a root library but the usual way to install deps is with go get and have github urls as imports.

kyle-github, 2022-07-02:
I am happy to set up a new Go repo in the libplctag project if you want to be the maintainer :-)

I don't program in Go at all and, unfortunately, the original contributor seems to have disappeared.  The wrapper is a bit out of date.   There is no support for the last few API additions such as the callbacks or string functions.

TheFern2, 2022-07-02:
I'd be happy to maintain the go wrapper, I don't program much in c but I'm sure I can figure out how to bring it up to the latest api.

kyle-github, 2022-07-02:
There shouldn't be much to do on the C side.   I can handle that.   I'll set up a new repo in the libplctag project and make you an admin for it.  Thanks!

kyle-github, 2022-07-02:
I set up a "goplctag" repo in the project.   I invited you to the admin group for it.   

I would love to get to the point where the CI process for the core library kicks off the build/test processes for the wrappers when I update it.   

Now if I could just find someone to take over the Java wrapper...

TheFern2, 2022-07-03:
Automated bindings is definitely the best way, there's swig for go and other langs. I'll give that a try. When the ci kicks in are there automated tests before pushing the release? or how's a new release pushed?

kyle-github, 2022-07-03:
I originally wrote the library to be used from Java, but I haven't used Java in my day job for years now.   So I hand wrote the original Java wrapper and did some small tweaks to make it a lot easier to use.   

The CI runs whenever there's a check-in on the prerelease or release branch.  Or a PR to either of those branches.  It doesn't push a new release.  I tag that manually after I do more testing.  Some of my tests run against my rack of PLC equipment so GitHub's CI can't do that testing.

I used Swig in some projects in the past, but it has been a long time.  The Java wrapper uses JNA which eliminates some of the build process that JNI requires.

TheFern2, 2022-07-03:
Ok makes sense, I will write some go tests before pushing new release, I also automate tests on my rack or VM. For the current wrapper would you say is fair to release AS IS assuming everything is backwards compatible which versions of the core do you think it supports? Once that's done, I can focus on swig bindings.

TheFern2, 2022-07-05:
I was doing some a visual inspection on the auto-generated bindings, and I noticed some functions with a generic name argument. Inspection of the core header file shows some of the int setters first argument with type and no identifier but the implementation does have an identifier.

Is this normal for c functions signatures? 

```
int plc_tag_set_int16(int32_t, int offset, int16_t val);
```

kyle-github, 2022-07-05:
It is legal but definitely an omission on my part.   I'll note this down to double check the .h file.

closed 2022-07-08 by TheFern2