libplctag/libplctag #375: Connect to OpenPLC Ethernet/IP soft-plc

pjbonestroo, 2022-07-18:
Before buying a Allen Bradley PLC I would like to use libplctag in my development environment to connect to a [OpenPLC](https://openplcproject.com/) device, which I can install as soft-plc. Do you think this is possible? Or what need to be done before this is possible? I tried it already (using libplctag by python) but I get errors when creating tags. I didn't dive into the errors yet, but first wanted to know from someone more experienced if I'm trying something feasible or not.

kyle-github, 2022-07-22:
You should probably raise this question in the [forum](https://groups.google.com/g/libplctag).   There are over 300 people that see those posts.

I do not use OpenPLC.   It looks interesting.

pjbonestroo, 2022-07-23:
@kyle-github Thanks, I've requested to participate in the forum.

closed 2022-07-28 by kyle-github