libplctag/libplctag #376: Control Logix PLC Connection Status

enesaygunea, 2022-07-27:
Hi, firstly thanks for this library. I have been developing almost 1 month in C++. 
I would like to learn how can I learn PLC Connection status? I want to learn is PLC connect? is PLC disconnect ? or etc. (any connection status) Because I want to learn connection quality.

C API provides plc_tag_status() function but I guess that does not work for me.
What is your suggestion?

kyle-github, 2022-07-28:
The [discussion you started on the forum](https://groups.google.com/g/libplctag/c/tC7uHuNGGn0) has more information.

kyle-github, 2026-01-13:
This should be in version 2.6.14.  Coming soon.

kyle-github, 2026-01-19:
This has been released.  Only 4 years late!  Thanks for reporting this.

closed 2026-01-19 by kyle-github