libplctag/libplctag #380: Problems trying to write strings (micro800)

GabrielRBr, 2022-08-23:
Hello gentlemen, I'm having some problems when using libplctag(1.0.33), I can't write more characters than the maximum number of characters already written previously.
For example if there was a string with 30 characters in memory, I can write up to this limit of 30.

I'm having trouble writing dynamic tags, and allocating more memory space.
Everything else here works normally, reading in micro800, writing/reading in Micrologix, ControlLogix, etc...

For the micro800 series is necessary some different treatment for writing strings?

kyle-github, 2022-08-31:
IIRC, Micro800 has strange strings.   I think they might be a single count byte followed by up to 255 character bytes.   Micro800 also does not support all of the CIP protocol commands that other AB PLCs do.  For instance you cannot list out the UDT definitions on a Micro800.   

The Micro800 series really is barely compatible with the rest of AB's product line.

Can you use the tag_rw2 tool to read a string and use the "--debug=4" command line argument to dump out a detailed log.   With that I can see what the strings look like to see if there's something we can try.

gotnone, 2022-10-21:
I had to insert `&str_count_word_bytes=1`  into the tag. The output is from the following command line:

`tag_rw2 --type=string --tag="protocol=ab-eip&gateway=192.168.1.246&plc=Micro800&elem_size=80&str_count_word_bytes=1&elem_count=1&name=Test4" --debug=4`

`Test4` is a `Controller Variable` according to CCW.  It is defined as a String of length 80 with contents `Nothing`.
list_tags defined it as:

`Tag "Test4" (00da) Counted character sting with 1 byte per character and 1 byte length indicator.  tag string = "protocol=ab-eip&gateway=192.168.1.246&plc=Micro800&elem_size=80&elem_count=1&name=Test4"
`

```
2022-10-21 12:29:14.095 thread(1) tag(0) INFO plc_tag_create_ex:833 Starting
2022-10-21 12:29:14.097 thread(1) tag(0) INFO initialize_modules:177 Starting.
2022-10-21 12:29:14.098 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
2022-10-21 12:29:14.098 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 12:29:14.098 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 12:29:14.098 thread(1) tag(0) INFO initialize_modules:207 Initialized library modules.
2022-10-21 12:29:14.098 thread(1) tag(0) INFO lib_init:135 Setting up global library data.
2022-10-21 12:29:14.098 thread(1) tag(0) INFO lib_init:137 Creating tag hashtable.
2022-10-21 12:29:14.098 thread(1) tag(0) INFO hashtable_create:75 Starting
2022-10-21 12:29:14.098 thread(1) tag(0) INFO hashtable_create:99 Done
2022-10-21 12:29:14.098 thread(1) tag(0) INFO lib_init:143 Creating tag hashtable mutex.
2022-10-21 12:29:14.098 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 12:29:14.098 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 12:29:14.098 thread(1) tag(0) INFO lib_init:149 Creating tag condition variable.
2022-10-21 12:29:14.098 thread(1) tag(0) DETAIL cond_create:1087 Starting.
2022-10-21 12:29:14.098 thread(1) tag(0) DETAIL cond_create:1114 Done.
2022-10-21 12:29:14.098 thread(1) tag(0) INFO lib_init:155 Creating tag tickler thread.
2022-10-21 12:29:14.098 thread(1) tag(0) DETAIL thread_create:884 Starting.
2022-10-21 12:29:14.110 thread(1) tag(0) DETAIL thread_create:917 Done.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO lib_init:161 Done.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO ab_init:123 Initializing AB protocol library.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO ab_init:132 Finished initializing AB protocol library.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO initialize_modules:215 Initializing Modbus module.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO mb_init:2410 Starting.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL mb_init:2412 Setting up mutex.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO mb_init:2421 Done.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO initialize_modules:226 Done initializing library modules.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO initialize_modules:231 Done.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=ab-eip
2022-10-21 12:29:14.111 thread(1) tag(0) INFO ab_tag_create:173 Starting.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from ab_tag_create:180
2022-10-21 12:29:14.111 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 0000021100255580
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL ab_tag_create:186 tag=0000021100255580
2022-10-21 12:29:14.111 thread(1) tag(0) INFO plc_tag_generic_init_tag:462 Starting.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL cond_create:1087 Starting.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL cond_create:1114 Done.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO plc_tag_generic_init_tag:493 Done.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL get_plc_type:907 Found Micro8xx PLC.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL ab_tag_create:246 Micro800 needs connected messaging.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL get_plc_type:907 Found Micro8xx PLC.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL session_find_or_create:235 Starting
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL session_find_or_create:259 Creating new session.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO session_create_unsafe:451 Starting
2022-10-21 12:29:14.111 thread(1) tag(0) INFO tag_tickler_func:508 Starting.
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL session_create_unsafe:454 Session should use connected messaging.
2022-10-21 12:29:14.111 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_unsafe:459
2022-10-21 12:29:14.111 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 000002110025a790
2022-10-21 12:29:14.111 thread(1) tag(0) DETAIL cip_encode_path:72 Starting
[... cut ...]
2022-10-21 12:29:14.160 thread(1) tag(0) DETAIL mutex_destroy:854 Done.
2022-10-21 12:29:14.160 thread(1) tag(0) DETAIL plc_tag_unregister_logger:1303 Starting
2022-10-21 12:29:14.160 thread(1) tag(0) DETAIL plc_tag_unregister_logger:1307 Done.
Library version 2.5.2.
Processing argument 1 "--type=string".
Setting type to TYPE_STRING.
Processing argument 2 "--tag=protocol=ab-eip&gateway=192.168.1.246&plc=Micro800&elem_size=80&str_count_word_bytes=1&elem_count=1&name=Test4".
Processing argument 3 "--debug=4".
data[0]="Nothing"

```

gotnone, 2022-10-21:
I had to insert `&str_count_word_bytes=1`  into the tag. The output is from the following command line:

`tag_rw2 --type=string --tag="protocol=ab-eip&gateway=192.168.1.246&plc=Micro800&elem_size=80str_count_word_bytes=1&elem_count=2&name=Test5" --debug=4`

`Test5` is a `Controller Variable` according to CCW.  It is defined as an array of length 2 of Strings of length 80 with contents [0] `Hello`, [1] `My Friend` .
list_tags defined it as:

`Tag "Test5[2]" (20da) Counted character sting with 1 byte per character and 1 byte length indicator.  tag string = "protocol=ab-eip&gateway=192.168.1.246&plc=Micro800&elem_size=80&elem_count=2&name=Test5"`

```
2022-10-21 13:04:46.226 thread(1) tag(0) INFO plc_tag_create_ex:833 Starting
2022-10-21 13:04:46.228 thread(1) tag(0) INFO initialize_modules:177 Starting.
2022-10-21 13:04:46.228 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
2022-10-21 13:04:46.228 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 13:04:46.228 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 13:04:46.228 thread(1) tag(0) INFO initialize_modules:207 Initialized library modules.
2022-10-21 13:04:46.228 thread(1) tag(0) INFO lib_init:135 Setting up global library data.
2022-10-21 13:04:46.228 thread(1) tag(0) INFO lib_init:137 Creating tag hashtable.
2022-10-21 13:04:46.228 thread(1) tag(0) INFO hashtable_create:75 Starting
2022-10-21 13:04:46.228 thread(1) tag(0) INFO hashtable_create:99 Done
2022-10-21 13:04:46.228 thread(1) tag(0) INFO lib_init:143 Creating tag hashtable mutex.
2022-10-21 13:04:46.228 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 13:04:46.228 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 13:04:46.228 thread(1) tag(0) INFO lib_init:149 Creating tag condition variable.
2022-10-21 13:04:46.228 thread(1) tag(0) DETAIL cond_create:1087 Starting.
2022-10-21 13:04:46.228 thread(1) tag(0) DETAIL cond_create:1114 Done.
2022-10-21 13:04:46.228 thread(1) tag(0) INFO lib_init:155 Creating tag tickler thread.
2022-10-21 13:04:46.228 thread(1) tag(0) DETAIL thread_create:884 Starting.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL thread_create:917 Done.
2022-10-21 13:04:46.248 thread(1) tag(0) INFO lib_init:161 Done.
2022-10-21 13:04:46.248 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
2022-10-21 13:04:46.248 thread(1) tag(0) INFO ab_init:123 Initializing AB protocol library.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 13:04:46.248 thread(1) tag(0) INFO ab_init:132 Finished initializing AB protocol library.
2022-10-21 13:04:46.248 thread(1) tag(0) INFO initialize_modules:215 Initializing Modbus module.
2022-10-21 13:04:46.248 thread(1) tag(0) INFO mb_init:2410 Starting.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL mb_init:2412 Setting up mutex.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 13:04:46.248 thread(1) tag(0) INFO tag_tickler_func:508 Starting.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 13:04:46.248 thread(1) tag(0) INFO mb_init:2421 Done.
2022-10-21 13:04:46.248 thread(1) tag(0) INFO initialize_modules:226 Done initializing library modules.
2022-10-21 13:04:46.248 thread(1) tag(0) INFO initialize_modules:231 Done.
2022-10-21 13:04:46.248 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=ab-eip
2022-10-21 13:04:46.248 thread(1) tag(0) INFO ab_tag_create:173 Starting.
2022-10-21 13:04:46.248 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from ab_tag_create:180
2022-10-21 13:04:46.248 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 00000266fa104510
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL ab_tag_create:186 tag=00000266fa104510
2022-10-21 13:04:46.248 thread(1) tag(0) INFO plc_tag_generic_init_tag:462 Starting.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL cond_create:1087 Starting.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL cond_create:1114 Done.
2022-10-21 13:04:46.248 thread(1) tag(0) INFO plc_tag_generic_init_tag:493 Done.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL get_plc_type:907 Found Micro8xx PLC.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL ab_tag_create:246 Micro800 needs connected messaging.
2022-10-21 13:04:46.248 thread(1) tag(0) DETAIL get_plc_type:907 Found Micro8xx PLC.
2022-10-21 13:04:46.249 thread(1) tag(0) DETAIL session_find_or_create:235 Starting
2022-10-21 13:04:46.249 thread(1) tag(0) DETAIL session_find_or_create:259 Creating new session.
2022-10-21 13:04:46.249 thread(1) tag(0) INFO session_create_unsafe:451 Starting
2022-10-21 13:04:46.249 thread(1) tag(0) DETAIL session_create_unsafe:454 Session should use connected messaging.
2022-10-21 13:04:46.249 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_unsafe:459
2022-10-21 13:04:46.249 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2022-10-21 13:04:46.249 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 00000266fa108430
[... cut ...]
2022-10-21 13:04:46.301 thread(1) tag(0) DETAIL mutex_destroy:854 Done.
2022-10-21 13:04:46.301 thread(1) tag(0) DETAIL plc_tag_unregister_logger:1303 Starting
2022-10-21 13:04:46.301 thread(1) tag(0) DETAIL plc_tag_unregister_logger:1307 Done.
Library version 2.5.2.
Processing argument 1 "--type=string".
Setting type to TYPE_STRING.
Processing argument 2 "--tag=protocol=ab-eip&gateway=192.168.1.246&plc=Micro800&elem_size=80&str_count_word_bytes=1&elem_count=2&name=Test5".
Processing argument 3 "--debug=4".
data[0]="Hello"

```

gotnone, 2022-10-21:
I can confirm that strings are limited to a maximum of 255 bytes in the CCW interface.  I played with various string options, but could not find a good combination which could read arrays of strings.  All strings on my micro850 needed the extra tag parameter `str_count_word_bytes=1`.

I think that the strings in an array may be concatenated with the length of the next string immediately following the first string.  I tried dumping the array as bytes.  See the output below.  The output shows that data[0] contains the first string length `5`.  data[1]:data[5] contain `Hello`. Then data[6] contains the next string length `9`. Finally, data[10]:data[15] contain `My Friend`.  This can be confirmed in the debug output at the bottom.

```tag_rw2 --type=uint8 --tag="protocol=ab-eip&gateway=192.168.1.246&plc=Micro800&elem_size=1&elem_count=2&name=Test5"
Library version 2.5.2.
Processing argument 1 "--type=uint8".
Processing argument 2 "--tag=protocol=ab-eip&gateway=192.168.1.246&plc=Micro800&elem_size=1&elem_co
unt=2&name=Test5".
data[0]=5 (5)
data[1]=72 (48)
data[2]=101 (65)
data[3]=108 (6c)
data[4]=108 (6c)
data[5]=111 (6f)
data[6]=9 (9)
data[7]=77 (4d)
data[8]=121 (79)
data[9]=32 (20)
data[10]=70 (46)
data[11]=114 (72)
data[12]=105 (69)
data[13]=101 (65)
data[14]=110 (6e)
data[15]=100 (64)
```

Full debug output
```
2022-10-21 13:31:00.994 thread(1) tag(0) INFO plc_tag_create_ex:833 Starting
2022-10-21 13:31:00.995 thread(1) tag(0) INFO initialize_modules:177 Starting.
2022-10-21 13:31:00.995 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
2022-10-21 13:31:00.995 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 13:31:00.995 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 13:31:00.995 thread(1) tag(0) INFO initialize_modules:207 Initialized library modules.
2022-10-21 13:31:00.995 thread(1) tag(0) INFO lib_init:135 Setting up global library data.
2022-10-21 13:31:00.995 thread(1) tag(0) INFO lib_init:137 Creating tag hashtable.
2022-10-21 13:31:00.995 thread(1) tag(0) INFO hashtable_create:75 Starting
2022-10-21 13:31:00.995 thread(1) tag(0) INFO hashtable_create:99 Done
2022-10-21 13:31:00.995 thread(1) tag(0) INFO lib_init:143 Creating tag hashtable mutex.
2022-10-21 13:31:00.995 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 13:31:00.995 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 13:31:00.995 thread(1) tag(0) INFO lib_init:149 Creating tag condition variable.
2022-10-21 13:31:00.995 thread(1) tag(0) DETAIL cond_create:1087 Starting.
2022-10-21 13:31:00.995 thread(1) tag(0) DETAIL cond_create:1114 Done.
2022-10-21 13:31:00.995 thread(1) tag(0) INFO lib_init:155 Creating tag tickler thread.
2022-10-21 13:31:00.995 thread(1) tag(0) DETAIL thread_create:884 Starting.
2022-10-21 13:31:01.006 thread(1) tag(0) DETAIL thread_create:917 Done.
2022-10-21 13:31:01.006 thread(1) tag(0) INFO lib_init:161 Done.
2022-10-21 13:31:01.006 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
2022-10-21 13:31:01.006 thread(1) tag(0) INFO ab_init:123 Initializing AB protocol library.
2022-10-21 13:31:01.006 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 13:31:01.006 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 13:31:01.006 thread(1) tag(0) INFO ab_init:132 Finished initializing AB protocol library.
2022-10-21 13:31:01.006 thread(1) tag(0) INFO initialize_modules:215 Initializing Modbus module.
2022-10-21 13:31:01.006 thread(1) tag(0) INFO mb_init:2410 Starting.
2022-10-21 13:31:01.006 thread(1) tag(0) DETAIL mb_init:2412 Setting up mutex.
2022-10-21 13:31:01.006 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2022-10-21 13:31:01.006 thread(1) tag(0) DETAIL mutex_create:753 Done.
2022-10-21 13:31:01.006 thread(1) tag(0) INFO mb_init:2421 Done.
2022-10-21 13:31:01.006 thread(1) tag(0) INFO initialize_modules:226 Done initializing library modules.
2022-10-21 13:31:01.006 thread(1) tag(0) INFO tag_tickler_func:508 Starting.
2022-10-21 13:31:01.007 thread(1) tag(0) INFO initialize_modules:231 Done.
2022-10-21 13:31:01.007 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=ab-eip
2022-10-21 13:31:01.007 thread(1) tag(0) INFO ab_tag_create:173 Starting.
2022-10-21 13:31:01.007 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from ab_tag_create:180
2022-10-21 13:31:01.007 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2022-10-21 13:31:01.007 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 00000232cacd5460
2022-10-21 13:31:01.007 thread(1) tag(0) DETAIL ab_tag_create:186 tag=00000232cacd5460
2022-10-21 13:31:01.007 thread(1) tag(0) INFO plc_tag_generic_init_tag:462 Starting.
[... 450 lines cut ...]
2022-10-21 13:31:01.059 thread(1) tag(0) DETAIL mutex_destroy:854 Done.
2022-10-21 13:31:01.059 thread(1) tag(0) DETAIL plc_tag_unregister_logger:1303 Starting
2022-10-21 13:31:01.059 thread(1) tag(0) DETAIL plc_tag_unregister_logger:1307 Done.
Library version 2.5.2.
Processing argument 1 "--type=uint8".
Processing argument 2 "--tag=protocol=ab-eip&gateway=192.168.1.246&plc=Micro800&elem_size=1&elem_count=2&name=Test5".
Processing argument 3 "--debug=4".
data[0]=5 (5)
data[1]=72 (48)
data[2]=101 (65)
data[3]=108 (6c)
data[4]=108 (6c)
data[5]=111 (6f)
data[6]=9 (9)
data[7]=77 (4d)
data[8]=121 (79)
data[9]=32 (20)
data[10]=70 (46)
data[11]=114 (72)
data[12]=105 (69)
data[13]=101 (65)
data[14]=110 (6e)
data[15]=100 (64)

```

kyle-github, 2022-11-05:
OK, so we know that there is a count byte followed by the character data (ASCII?).

In order to try to write a string, you will need to make sure that the tag buffer is long enough (and this still might not work).

Here is a quick sketch in C.   I haven't compiled this so read as if this was pseudo-code.

```
    // read in the string.
    rc = plc_tag_read(tag_id, 1000);

    char str_buf[256] = {0};
    rc = plc_tag_get_string(tag_id, 0, &str_buf[0], (int)(unsigned int)sizeof(str_buf));

    printf("string: %s\n", str_buf);

    // now make the string bigger.
    const char *new_str_val = "Good morning, Vietnam!";
    int new_len = strlen(new_str_val);

    rc = plc_tag_set_size(tag_id, new_len+1);  // +1 for the count byte.

    rc = plc_tag_set_string(tag_id, 0, new_str_val);

    rc = plc_tag_write(tag_id, 1000);
```

Again, this might not work if the PLC does not allow strings to be resized or does not have some internal larger buffer it uses.   Control/CompactLogix allocate 88 bytes per string no matter how long it is.  So you can write to them with any length up to 82 characters.

kyle-github, 2022-11-05:
Hmm, I just noticed that at the top of these comments you said that the test string tag was of size 80.   So that might be the limit that you can write (or maybe 79).

gotnone, 2022-11-06:
The default string length in CCW is 80 characters.  You can go up to a maximum of 255 characters.  The total length is the string length selected in CCW + 1 byte for the string count word (byte) which precedes the string contents.

I have not verified the entire character set, but initial tests seem to be ASCII.  I tried to place non ASCII (UTF8) characters in the string from CCW and it gave me back `?`s (0x3f) for each character in both the tag results and in the CCW variable browser.  It converted `€`  to `Ç` 0x80 and `£` to `ú` 0xa3 in the tag output, but the CCW variable browser shows the Euro and Pound symbol correctly.

I don't know if or how non ASCII characters could be used.  Perhaps users with a different default character set could write single byte characters from that character set to a tag and output the same.

kyle-github, 2022-11-13:
If the write test above works, then I can see if I can modify the generic PLC code to increase the size of the tag buffer in memory if there is one string (no arrays).

Can you try to run that snippet of code above?

gotnone, 2022-11-14:
It does not seem to work correctly.  I have attached the log file at debug level 4.

At the start of the program the tag containing the string "Nothing".  After attempting to write to it, it still contains "Nothing".  I still need to add the extra parameter `str_count_word_bytes=1` to the tag path.

[debug.log](https://github.com/libplctag/libplctag/files/9999309/debug.log)

kyle-github, 2022-11-14:
Thanks for the logs.   It looks like there is a check in my code somewhere that is still failing.   I'll have to try to recreate this here to figure out what is going on.

```text
2022-11-13 21:40:29.799 thread(1) tag(11) WARN plc_tag_set_string:3460 string_capacity=82, string_last_offset=83, tag_size=23.
2022-11-13 21:40:29.799 thread(1) tag(11) WARN plc_tag_set_string:3472 Bad configuration? String capacity/size is larger than the tag buffer!
```

kyle-github, 2022-11-15:
Ah, OK.   It looks like you need to set `str_max_capacity` as well.  If that is not set then the code checks for the existing string length.  I would start with 80 and see if that works.