libplctag/libplctag #384: Read/Write Omron CJ2M PLC Strings

kyle-github, 2022-10-01:
Here's the deets from @jfontsaballs:

- Strings are up to 255 bytes (does this include the count word?).
- Strings have a count word of two bytes.   Little endian.
- Strings are padded to a multiple of 2 bytes in length with zero bytes, but there is no guaranteed zero termination byte!
- String bytes are in order.

So we need defaults of:

str_count_word_bytes=2
str_is_byte_swapped=0
str_is_counted=1
str_is_fixed_length=0
str_is_zero_terminated=0
str_max_capacity=255
str_pad_bytes=0
str_total_length=0

The padding to multiples of 2 bytes is new, so we will need to add that functionality and a new attribute.

kyle-github, 2022-10-01:
Can we have arrays of strings?

Are strings variable length in memory?

kyle-github, 2022-10-01:
Does this only apply to the CJ2M family?  What about the NX/NJ?

jfontsaballs, 2022-10-01:
Next week I'll double check these questions with actual hardware and post the answers.

I must warn you that right now I only have access to a CJ2M PLC, however in the coming months I may have access to an NX1P2. If I find any differences from the CJ2M I'll post them in due course.

jfontsaballs, 2022-10-03:
These has been verified on the CJ2M family only.

In the PLC's memory strings are allocated a fixed length up to 255 characters and they are null terminated. They can not be formed into arrays.

Strings must be of even length when sent over Ethernet/IP.

Dump of writing an odd length string with a single "a" (TCP headers omitted):
```
0000   6f 00 32 00 00 15 02 00 00 00 00 00 1a 00 00 00
0010   00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00
0020   00 00 00 00 b2 00 22 00 52 02 20 06 24 01 06 9c
0030   14 00 4d 05 91 08 4d 79 53 74 72 69 6e 67 d0 00
0040   01 00 02 00 61 00 01 00 01 00
             -----------
```

Dump of writing an even length string with two a's "aa" (TCP headers omitted):
```
0000   6f 00 32 00 00 15 02 00 00 00 00 00 1c 00 00 00
0010   00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00
0020   00 00 00 00 b2 00 22 00 52 02 20 06 24 01 06 9c
0030   14 00 4d 05 91 08 4d 79 53 74 72 69 6e 67 d0 00
0040   01 00 02 00 61 61 01 00 01 00
             -----------
```

kyle-github, 2022-10-04:
Thanks for this data!   

Can you have arrays of strings and write them on an Omron?   With AB, you generally can because STRING takes up a fixed amount of space.  However, if you have variable size strings, then I am not sure what the PLC is going to do.

jfontsaballs, 2022-10-06:
CJ2M cannot have arrays of strings.

jfontsaballs, 2022-10-12:
I don't know how big a single message can be, and I don't have tested it. The 256 bytes limit refers to the maximum string length on legacy PLCs, it does not have anything to do with the Ethernet/IP maximum length. Also I believe the newer PLCs do not have this limit upon string length.

jfontsaballs, 2022-10-12:
CJ2M supports the FINS protocol. Newer PLCs (NX/NJ) still support it for compatibility with older hardware (mainly NS HMIs) but consider it legacy and require special actions to make PLC variables available through this protocol.

Moreover the FINS protocol is based on memory addresses (similar to Modbus) which makes it less practical to use than Ethernet/IP.

For these reasons I recommend against the FINS protocol.

jfontsaballs, 2022-10-12:
Omron Sysmac software has a trial version available for download [https://automation.omron.com/en/us/forms/download-sysmac-studio-trial].

NX1P2-9B24DT is the less powerful (and probably cheapest) Omron PLC compatible with Sysmac software. It has support for Ethernet/IP. In these times of scarcity of electronic components the NPN versions seem to be more readily available than the PNP ones. For what I know it's usually much cheaper than any Rockwell PLC.

Be careful that CP2E PLCs (which are quite cheap), although they have an Ethernet port, they are not compatible with the Ethernet/IP protocol. I believe that the only legacy (CX-Programmer) PLCs compatible with this protocol are the CJ ones which are much more expensive and seem to be deprecated.

kyle-github, 2022-10-13:
Looks like there is a small amount of documentation here: https://assets.omron.eu/downloads/manual/en/w506_nx_nj-series_cpu_unit_built-in_ethernet_ip_port_users_manual_en.pdf

Not a lot :-(

kyle-github, 2022-11-13:
Hi @jfontsaballs, I am back looking at this.  I was ready to implement something here and realized that there is information that I missed.   What comes back when an odd-length string is read?   Is the zero padding there?  I was happily writing code with the assumption that the strings were the same read and written and then decided that that is a bad assumption.

Can you do a capture of reading of an even length string and a capture of reading an odd length string?

jfontsaballs, 2022-11-15:
I'm very sorry but right now I don't have a PLC available to gather the samples. In any case, thank-you very much for your work!

zN3utr4l, 2024-03-13:
I have an Omron PLC at the moment, as you can understand from the open issues, I could test it myself if there is some code to test