libplctag/libplctag #385: cli does not compile on mingw-w64

gotnone, 2022-10-10:
Mingw-w64 does not include getline() even when the feature test macro -D_POSIX_C_SOURCE=200809L is enabled.  This function is missing in both msvcrt and ucrt.  Until this is fixed, Mingw-w64 needs to have the contents of the src/contrib/cli/getline.h enabled.
```
diff --git a/src/contrib/cli/getline.h b/src/contrib/cli/getline.h
index 41789b9..b56c130 100644
--- a/src/contrib/cli/getline.h
+++ b/src/contrib/cli/getline.h
@@ -1,4 +1,4 @@
-#ifdef _MSC_VER
+#if defined(_MSC_VER) || defined (__MINGW64__)
     #define _CRT_SECURE_NO_WARNINGS 1
     #define restrict __restrict
```

kyle-github, 2022-10-10:
Thanks!  I'll try to get this into the next release.

kyle-github, 2022-11-04:
Fixed in v2.5.3.  Sorry for the delay.

closed 2022-11-04 by kyle-github