libplctag/libplctag #386: The process cannot access the file 'plctag.dll'

avnet78, 2022-10-12:
I am getting the below exception while writing to an EthernetIP device:

Status: The process cannot access the file 'plctag.dll' because it is being used by another process. 

Please advise.

kyle-github, 2022-10-13:
Thanks for reaching out.  I need to ask some questions before I can try to help.

From the name, are you using Windows?   If so, what version of Windows?   

Are you using one of the wrappers like the .Net wrapper?  If not, how are you installing the system?

avnet78, 2022-10-13:
Thank you very much for your quick response.

Yes, I am using the libplc.net wrapper. The source code build is deployed on windows 11 server. I’m using dotnet framework version 452.
I have two windows  services or deamon processes.  One reads values from the ethernet IP device and the other one writes values to the ethernet IP device.  Please note, both processes are running simultaneously.
I don’t have any issues with the reads but the writes are throwing exception saying that the DLL is being used by another process.
Please let me know if you have any questions or need more information.

Thanks,
Vishal

kyle-github, 2022-10-13:
Ah, you need to reach out to Jody and Tim on the [libplctag.Net issues list](https://github.com/libplctag/libplctag.NET/issues).  They can probably help you.   I know that they have worked with other users on similar problems in the past.   It sounds like something is happening with permissions or perhaps during install to me.

avnet78, 2022-10-13:
Thank you Kyle. I created an issue in the libplctag.Net issues list<https://github.com/libplctag/libplctag.NET/issues>. Could you please ask Jody & Tim to look into it?

kyle-github, 2022-10-14:
Looks like Tim already responded.  Thanks, Tim!

closed 2022-10-14 by kyle-github