libplctag/libplctag #388: Documentation on accessing boolean values

timyhac, 2022-10-14:
One of the more common issues posted to libplctag.NET is variations on how to access boolean values.

Boolean values can be represented in PLCs in a myriad of different ways, requiring use of different access methods.

This issue requests that documentation is developed that shows what these different access methods are.
libplctag users (e.g. libplctag.NET) could base their access on this documentation.

The way I see this being done is a big table of combinations of parameters and how the tag should be accessed.

- PlcModel
- Protocol
- Type (e.g. bool tag or bool array tag or bit within integer, or UDT field, etc)

We should be able to enumerate the entire set of combinations, and when we get issues, we can refer library consumers to this table. Some parameter combinations we won't have answers to, and it will provide an opportunity for library consumers to contribute.

timyhac, 2022-10-14:
I will have a crack at this but I no longer have access to hardware so I can only revisit old github issues and google gropus threads for info.

kyle-github, 2022-10-14:
This sounds like a good idea.   I know I have sent out emails and forum threads on this, but maybe it should be pulled into a Wiki.   I'll see what I can come up with.

On a slightly different topic, I wonder if there should be a "non-denominational" wiki that is for the core library, the .Net library etc.   I wrote up a wiki page on string handling because it is such a complicated topic, but it applies to the core library and (with some translation) to the wrappers too.

timyhac, 2022-10-15:
In terms of the non-denominational wiki, I agree 100% - my preference for documenting libplctag.NET is to mostly defer to the core documentation, and just document the differences. The differences should just be usage patterns as opposed to functional behaviour differences - @jkoplo and I discussed this recently and decided that improving the examples is the way to go.

One thing I recall needing to unlearn is that when using libplctag, you're actually working with two distinct interfaces - the libplctag API, and the device API. Not sure if it would be better to create yet another page that talks about this or not. Probably sticking to the boolean access I guess would cover a lot of it.

kyle-github, 2022-11-05:
Finally got a first cut up [here](https://github.com/libplctag/libplctag/wiki/Bits-and-Booleans).

timyhac, 2022-11-08:
This is great, I did not know about the tag named based bit access

kyle-github, 2022-11-11:
Close for now.   Will reopen or just edit the wiki page.

closed 2022-11-11 by kyle-github