libplctag/libplctag #391: Get raw bytes signature

timyhac, 2022-10-16:
A common pattern when using libplctag.NET is to do something when the value of a tag changes.
The things people typically want to do are either record the change or update the UI.
For simple values this is straightforward, you can simply compare the values after each read.
For UDTs and Arrays this is less performant/straightforward and typically would involve generating a signature of the buffer with some hashing function .

What do y'all think about exposing that in the C API?

```c
// Something like this
int plc_tag_get_raw_bytes_signature(int32_t tag_id);
// returns one of the STATUS_CODES or a positive value representing the signature
```
I guess the benefits would be
1/ Performance - don't need to transfer buffer to managed land unless required.
2/ Common use case - so don't need to ask library consumers to roll their own.

I'm not sure what the appropriate hashing function would be, or if there would be different choices for hashing functions in different scenarios, or what the ramifications of hash collisions would be...

```c
// Or if it would be useful to get signature for only part of the buffer
int plc_tag_get_raw_bytes_signature(int32_t tag_id, int start_offset, int hash_length);
```

kyle-github, 2022-10-16:
Very interesting idea.

The purpose of this would be to trigger a "value changed" event, right?   At least that seems to be the driving use case here.   That lets people (especially with some wrapper magic) have relatively friction free change events.

I like the idea in concept, but I am wondering at the cost.  Calculating a hash or doing a comparison would be a possibly expensive operation.   I wouldn't want to do it in all cases due to the CPU impact (imagine what doing this every 50ms for a 1000 element DINT array would cost!).   At the same time, it would be low friction to have some way to generate a changed event and use the existing callback structure.

Maybe we can take this apart a bit.   What if we have a combination of things:

1. We use the auto-sync tag read feature to read the value in periodically.
2. We add a new event value, PLCTAG_EVENT_VALUE_CHANGED.
3. We add a new API function that lets you register a C function that will calculate a changed value: `int plc_tag_check_register_change_check_function(int32_t id, int (change_check_func)(int32_t id, void *context), void *context)`.   
4. When there is a function registered, we will call it after every successful read and if it returns 1, we will emit a PLCTAG_EVENT_VALUE_CHANGED event.   If the user has an event callback registered then it will be called for that.

I added a context argument so that you could roll your own comparison.  For instance if you really wanted to catch everything, you could copy the contents of the tag and do a byte-by-byte comparison.   The comparison function only has to return 1 on change, 0 on no change or any of the error codes (all negative) when there is a problem.

This might be too much effort on the part of the library user.   If we fall back to your original idea, then there isn't really any difference between getting the hash from the C code and just reading the raw bytes on every read event and calculating your own hash.   Collisions are a serious issue unless you use one of the cryptographic hash functions and those are not cheap.   I wouldn't want to calculate SHA256 on an embedded system for every read :-(

Part of me says just write an internal implementation and simply emit the CHANGED event all the time and let people catch it if they want.   Ugh, the CPU and memory cost though...

I really like this idea but I want to refine it a bit to see if we can find a solution that has very low coding cost for users but is not too expensive in CPU and works in the vast majority of the common cases.  I think we need to develop a good driving use case for this and see if it would solve most of the problems.

kyle-github, 2022-10-16:
Hmm, another thought that is a bit more expressive.   What if we allow a function to be registered that gets called with events first and then turns around and generates its own events for the user-callback to handle.

```c
int plc_tag_register_event_filter(int32_t id, void (*tag_callback_func)(int32_t tag_id, int event, int status, void *userdata), void *userdata);
int plc_tag_unregister_event_filter(int32_t id);
```

Add a new event type PLCTAG_EVENT_NO_EVENT.   If the interceptor function returns that, then do _not_ call the user callback.   If the interceptor returns anything else at all, use that value for the event and call the user callback.

You could use the interceptor as a filter to only pass on things you care about, or to generate changed events, or to translate events from libplctag's values to something else.

I'm probably overthinking this.

timyhac, 2022-10-16:
Yes that is the key use case. Is there a nuance where you might want to compare the value in between instances of an application? In that case libplctag won't have an opportunity to emit a changed event, but the consuming application could record the signature before shutdown and check again on startup. Not sure if this is important.

Had a thought: in terms of which hash algorithm to use, we could ask the consumer to pass their selection through as an attribute:
e.g. hash_algorithm=crc32|md5|sha256|fnv1
This would encourage thoughtful selection of appropriate hashing function.

Or it could be in the function call itself (which would _force_ thoughtful selection).
```c
// enum hash_algorithm { crc32 = 1, md5 = 2, sha256 = 3, fnv1 = 4 };
int plc_tag_get_raw_bytes_signature(int32_t tag_id, int hash_algorithm, int start_offset, int hash_length);
```

With the event filter idea, wouldn't you lose any performance benefits? It would be the same as listening to ReadCompleted? At least for managed languages where you have the blip overhead marshalling between native and managed contexts.
Absolutely it is more expressive though.

timyhac, 2022-10-16:
> there isn't really any difference between getting the hash from the C code and just reading the raw bytes on every read event and calculating your own hash.

Absolutely.

Maybe this is more about finding ways to minimize the overhead in the wrapper.

kyle-github, 2022-10-18:
One of the problems with having a built-in hashing function is that most of them require fairly significant state.   I try to keep the overhead of each tag as small as I can but a typical tag still has more than 200 bytes of overhead.   That could double with some of the hash functions out there.

timyhac, 2022-10-18:
Why would it double the size of each tag?
My guess at an implementation of the get_signature function would have been a static function.

kyle-github, 2022-10-19:
Yeah, I can't use static buffers and if you look at the [Wikipedia cryptographic hash function comparison](https://en.wikipedia.org/wiki/Comparison_of_cryptographic_hash_functions), you can see that most have pretty large state.   That state has to be in the tag itself or on the stack so that it can be mutex protected.   There are two parts to the state.  The first is the hash result.  We'd need to save that so that we can do a comparison.  The second is the intermediate state that is used.   That gets large too.   We could choose some functions that don't take too much room such as SHA-1.   That's not too big with only 20 bytes for the output and 20 bytes for the internal state needed during calculation.  But SHA-1 isn't that great.  Better than RC4 or MD5 though.

We probably cannot just return a 32-bit int and call it enough.   A lot of things break surprisingly often with only 32 bits.

timyhac, 2022-10-19:
Ah I see - good points! Looks like it's a no-go and the appropriate place for value change detection is the consuming application/code.

kyle-github, 2022-10-19:
Let's keep this open to keep thinking about it.   40 bytes of state for SHA-1 is not completely horrible.   Perhaps some spin on my first proposal where I broke down the idea a bit to allow user code to do the calculation but provided a hook for it.  Or maybe the event filter idea.

I agree that the overhead from a managed language like Java or C# to C and back might be a problem.   Do you have a good way of measuring that?  The JNA library I used for Java does a lot of stuff at runtime so it has a lot of overhead.

I think if we can:

1. make the cost only apply to specific tags,
2. have a good default implementation in the C library,

we can have something that works for everyone.  I know that OPC UA can do this so perhaps I should look at what that does.

timyhac, 2022-10-20:
> overhead ... managed language ... might be a problem. Do you have a good way of measuring that?

I'll have a think about this.

My first thought would be to compare the cost of calling `plc_tag_get_raw_bytes(...)` through P/Invoke, with the cost of copying a reference to a buffer purely in managed land.
[The interop declaration says that the buffer is blittable](https://github.com/libplctag/libplctag.NET/blob/9122447b3ca2fd6c5a8bc5fbd62c177a4036bf4d/src/libplctag.NativeImport/NativeMethods.cs#L219), so the difference should be the cost of whatever P/Invoke does.
I don't know if these assumptions are valid.
I'll try it anyway.

HOWEVER

This thought stream illuminates an issue with how [`GetBuffer()` is currently exposed](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/NativeTagWrapper.cs#L476) in the high-level wrapper: It instantiates a new buffer for the core library to fill _every_ time its called. So its not just about the  native<->managed thunk, but also the cost of garbage collecting all of these buffers.
I think an action on libplctag.NET would be to expose an overload `GetBuffer(byte[] buffer)` - given that the consumer has more context about how the buffer is to be used, it could potentially reuse the same buffer between successive calls.

timyhac, 2022-10-20:
One advantage of the filter proposal is that the meaning of "ValueChanged" could be context-specific, particularly when talking about floating point data. Asking the user to supply their own filter would allow them to supply custom logic (e.g. a hash algorithm would assume +0 and -0 are different values, but applications might not care). Would this require two copies of the buffer (old/new)?
Although if there was a first-pass that used hash detection, and then a second pass that did more context-specific change detection, it could save some CPU cycles.

kyle-github, 2022-10-21:
Hmm, I hadn't thought of that.   You could also potentially combine multiple BOOL fields of an ADT to raise something like a alarm-raised event.

How bad is the overhead of a callback when interoperating from C to .Net?

kyle-github, 2022-10-22:
The more I think of it, the more I like the filter idea.  But...

I keep coming back to a core question: is a filter really unique enough to justify a separate new API function and the overhead?  There's nothing there that cannot be done with the existing callback system.    The counter argument is that having a "changed" event is a really, really common and useful pattern in programming.  So are filters.

What I should have done, in retrospect, was have an ability to chain callbacks and to have callbacks indicate whether they were the end of the chain or not for a given event.

To make this testable, I need to implement a new kind of tag or add to the system tag type.  This tag type needs to be a dummy tag that can be written and read, with auto-sync too.   It needs to generate all the normal events.

So here's a plan:

- [ ] implement a dummy tag type.
- [ ] determine the filter function API.
- [ ] implement the filter function API.
- [ ] test, test, test.

The first one can be released independently and will help with some of the CI and other tests.   It will also allow for a lot more testing for wrappers (since you won't need a PLC).

kyle-github, 2022-10-30:
I keep working through possible ways to implement a simple PLCTAG_EVENT_CHANGED_VALUE event...   Even if I implement the most highly optimized (AVX2 on Intel) CRC32 algorithm, I don't think I can afford to do it per tag.   When I do some simple tests, the CPU overhead is pretty bad.

I limit the number of threads to about one per PLC.   That works because 90% of the work is network IO and that doesn't take a lot of CPU time.  With just calculating a CRC32 value, this jumps significantly if I do it for all tags.

I guess this says that doing this for all tags is off the table.   It has to be something like either the original proposal or the filter proposal where it is only on the tags that the application wants and is willing to pay for.

kyle-github, 2022-11-23:
Hey @timyhac, where do we want to go from here?   I think the whole issue exposes that I should have had stacking event handlers from the start instead of just one.

timyhac, 2022-11-23:
It was a useful exercise to explore these ideas - to me it sounds like the cost/benefit equation doesn't stack up to proceed.

timyhac, 2022-11-23:
Discovering the need for a dummy tag type was useful in itself I think

kyle-github, 2025-03-04:
I think this is fixed, no?   Please reopen if not.   I know I added the raw data type attribute.

closed 2025-03-04 by kyle-github

timyhac, 2025-03-04:
I believe we decided against providing a built-in function for providing a hash signature of the raw bytes. There hasn't been any request from libplctag.NET for this and even if there was its likely that [providing a Span<byte> based overload for GetBuffer](https://github.com/libplctag/libplctag.NET/issues/423) would be suitable for this performance use case. Thanks for the discussion!