libplctag/libplctag #396: Can not compile C++ example.

AsifArain, 2022-11-08:
I am trying to compile C++ example as `g++ .\simple_cpp.cpp -o simple_cpp.exe` but getting the following errors. Am I missing some links/paths? 

C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x2d): undefined reference to `__imp_plc_tag_check_lib_version'
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x86): undefined reference to `__imp_plc_tag_create'
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x9d): undefined reference to `__imp_plc_tag_decode_error'     
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0xd8): undefined reference to `__imp_plc_tag_status'
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0xf4): undefined reference to `__imp_plc_tag_decode_error'     
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x125): undefined reference to `__imp_plc_tag_destroy'
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x142): undefined reference to `__imp_plc_tag_read'
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x159): undefined reference to `__imp_plc_tag_decode_error'    
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x190): undefined reference to `__imp_plc_tag_destroy'
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x1bf): undefined reference to `__imp_plc_tag_get_int32'       
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x212): undefined reference to `__imp_plc_tag_get_int32'       
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x260): undefined reference to `__imp_plc_tag_set_int32'       
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x279): undefined reference to `__imp_plc_tag_write'
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x290): undefined reference to `__imp_plc_tag_decode_error'    
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x2c7): undefined reference to `__imp_plc_tag_destroy'
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x2e4): undefined reference to `__imp_plc_tag_read'
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x2fb): undefined reference to `__imp_plc_tag_decode_error'
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x332): undefined reference to `__imp_plc_tag_destroy'     
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x35e): undefined reference to `__imp_plc_tag_get_int32'   
C:\Users\SEARAINA\AppData\Local\Temp\ccwlC16T.o:simple_cpp.cpp:(.text+0x39a): undefined reference to `__imp_plc_tag_destroy'     
collect2.exe: error: ld returned 1 exit status

kyle-github, 2022-11-09:
You need to link in the libplctag library.   You'll have to check the exact command line but it is probably something like:

```text
gcc simple_cpp.cpp -o simple_cpp -lplctag 
```

Do you have the library files (DLL and LIB) and the include file (libplctag.h) and the source file in the same directory?

AsifArain, 2022-11-09:
Sorry for my limited knowledge, but where can I find libplctag library? 
I have downloaded the `libplctag-release` and tried `g++ simple_cpp.cpp -o simple_cpp -lplctag` in `~\libplctag-release\src\examples>` directory, `libplctag.h` in `~\libplctag-release\src\lib` directory and I am still getting the following error:

> c:/mingw/bin/../lib/gcc/x86_64-w64-mingw32/9.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lplctag
collect2.exe: error: ld returned 1 exit status

`libplctag.h` header file is included in `simple_cpp.cpp` as `#include "../lib/libplctag.h"`, so I believe it should work.

kyle-github, 2022-11-10:
There is not a pre-built release for MinGW.   You will need to build it yourself if you want to use that system and compiler.   Is there a reason you need MinGW?   There are versions of Visual Studio that are free.   And there is WSL as well.

I do not use MinGW but others have contributed fixes to make it work.  I do not know if it can use the binaries provided by Visual Studio (which is what I distribute for Windows).

kyle-github, 2022-11-10:
The [build docs](https://github.com/libplctag/libplctag/blob/release/BUILD.md) on the site explain what to do to build the entire project.  That includes building simple_cpp.cpp as part of that.   Then you can set up a separate Visual Studio project with the libplctag libraries and header file.

kyle-github, 2022-11-23:
Please reopen if you are still having problems or ask on the [forum](https://groups.google.com/g/libplctag).  There are several hundred people on the forum and it is likely that someone will be able to help!

closed 2022-11-23 by kyle-github