libplctag/libplctag #400: How much is considered thread safe?

Laro88, 2022-11-14:
Got a multithreading question.

I have multiple threads accessing some tags (mostly read operations) and I tried to find some documentation stating how much (if not all) that is threadsafe to use.

My reason is that I might have seen a write operation that messed up the value read almost simultaneously from another thread. I don't have access to the PLC program, but the PLC programmer assured me that the value I observed was not possible to get. 

I am on 2.3.5 (on Windows Server 2019 unfortunately) at the moment but I hope to upgrade in a not to distant future.

Besides this libplctag has been running flawlessly for at least 1E7 read ops and some writes) before the system was to be turned of :-)

kyle-github, 2022-11-14:
The library is designed to be fully threadsafe within a single PC.   That said, the PLC might not be.  I have seen cases where the writes to the PLC are not coordinated with the program scans or other activity such as produced tags.   In these cases, you can see larger writes (more than one atomic object such as a DINT) be partially completed when the program scan happens.

We worked around this by having sentinel values at the start and end of larger writes.   We just used DINT values and incremented them monotonically.  First, we updated the first one, then the rest of the data, then the last one then we would write.   In the PLC we changed the code to look at the sentinel values.  If they matched, then the data was good.  If they did not match, then the data was partially written and the processing was skipped.

If you are seeing situations where writes are immediately followed by reads, it is possible that the write is not fully complete in the PLC.   Normally this would not be possible or likely as the library maintains a single connection to each PLC (unless you force it to do otherwise).  This single-threads all access to the PLC.   However, it is possible for requests to be packed into one larger request and sent to the PLC all at once.   So, it is possible for a write and a read to be sent to the PLC at the same time.   The PLC processes these requests in some sort of order, but I do not know what that is.

If you want to coordinate outside the API calls between threads, you can use the `plc_tag_lock()` and `plc_tag_unlock()` calls.  Have the write thread lock before it starts writing, and only unlock when the write is fully complete.  Have the read thread do the same.  This will ensure that the full round-trip request-response is complete before the next operation occurs.   Normally, if you let the library do the waiting (synchronous mode) you'll get the same effect.

kyle-github, 2022-11-23:
Please reopen if you have further questions or you think you found a case where the library is not working correctly!  I am _VERY_ interested in bug reports!

closed 2022-11-23 by kyle-github