libplctag/libplctag #408: Listing tags on Micro800 does not show BOOL arrays

kyle-github, 2022-12-15:
Just what it says on the tin.   Listing out tags on a Micro800 shows other array types, but leaves out any BOOL array tags.  See the [original issue](https://github.com/libplctag/libplctag.NET/issues/311) in the .Net wrapper.

This appears to be a bug in the Micro800, but hopefully some solution can be found.

kyle-github, 2023-01-02:
Adding tags to show that this is considered a bug, but it is also an enhancement.   Testing by users of pylogix show that the Micro800 is _NOT_ returning boolean arrays in the list of tags.   More tests are going to be run when they get a test Micro800.

gotnone, 2023-09-06:
If you have some specific tests I have access to a micro800

kyle-github, 2025-04-11:
The tests by the Pylogix users show that this is definitely not working on the PLC side.   Or if it is, there is some other CIP calls that must be used.

@gotnone if you still have access to a Micro800, can you run Wireshark and bring up CCW to see what it does to find the arrays?   I suspect that there is something we are all missing.  It is definitely not documented.

gotnone, 2025-04-12:
I am not exactly sure what I am looking for.  Do you want the text dissections or the full packet captures.  Running a diff on the text captures after deleting all the columns up to `CIP` did show some differences.

[image]

[image]

Dissections
[CCW_Dump_NO_Array.txt](https://github.com/user-attachments/files/19717410/CCW_Dump_NO_Array.txt)
[CCW_Dump_Bool_Array.txt](https://github.com/user-attachments/files/19717411/CCW_Dump_Bool_Array.txt)

kyle-github, 2025-04-12:
This is quite helpful!   Can you get the full capture with Wireshark?  Knowing the name might help me look through the captures.  There is clearly something special about class 0x350 and the File class.

gotnone, 2025-04-13:
@kyle-github I have a few captures.  

### Baseline Capture with no Bool Arrays

This first one only has a regular Bool exported in the Global Variable section

[CCW_Micro850_NOArray_CIP.pcapng.gz](https://github.com/user-attachments/files/19722611/CCW_Micro850_NOArray_CIP.pcapng.gz)

### Single Bool Array capture

The second one is as only one Bool Array Exported

[CCW_Micro850_BoolArray_CIP.pcapng.gz](https://github.com/user-attachments/files/19722628/CCW_Micro850_BoolArray_CIP.pcapng.gz)

### Many Bool Arrays capture

This last one has many different Bool Arrays Exported

[image]

[image]

I also copied some magic values into the Bool Arrays to help locate them in the capture

| TAG | Value 1 | Value 2 |
|-----|----------|--------|
| TEST_16_BA | 0x5555 | 0xAAAA |
| TEST_32_BA | 0xDEADBEEF | 0xBAADDECA |
| TEST_64_BA | 0xAAAABBBBCCCCDDDD | 0x5555AAAA5555AAAA |
| MYBOOLARRAY | 0xABCDFEDC |  0xFADECAB0 |

For MYBOOLARRAY, I only copied in the lowest 25 bits.  All the other BoolArrays were a power of 2 in size.  I was having a hard time getting variable bit indexing to work in the structured text block, so I manually copied the bits.  It may be that my hardware doesn't support the latest updates. It is EOL and only supports project version 12 (the active SKUs are up to project version 22).
```
BoolArr[0] := In.0;
BoolArr[1] := In.1;
BoolArr[2] := In.2;
BoolArr[3] := In.3;
BoolArr[4] := In.4;
BoolArr[5] := In.5;
BoolArr[6] := In.6;
BoolArr[7] := In.7;
BoolArr[8] := In.8;
BoolArr[9] := In.9;
BoolArr[10] := In.10;
BoolArr[11] := In.11;
BoolArr[12] := In.12;
BoolArr[13] := In.13;
BoolArr[14] := In.14;
BoolArr[15] := In.15;
BoolArr[16] := In.16;
BoolArr[17] := In.17;
BoolArr[18] := In.18;
BoolArr[19] := In.19;
BoolArr[20] := In.20;
BoolArr[21] := In.21;
BoolArr[22] := In.22;
BoolArr[23] := In.23;
BoolArr[24] := In.24;
```

TEST_1_BA[0] was used to toggle between output values for the other Bool Array variables.  TEST_2_BA was left unmodified in the code.  It should have 0 values.

[image]

[image]

[CCW_Micro850_Many_BoolArrays_CIP.pcapng.gz](https://github.com/user-attachments/files/19722742/CCW_Micro850_Many_BoolArrays_CIP.pcapng.gz)

gotnone, 2025-04-13:
In the [CCW_Micro850_Many_BoolArrays_CIP.pcapng.gz](https://github.com/user-attachments/files/19722742/CCW_Micro850_Many_BoolArrays_CIP.pcapng.gz) capture you can see the magic numbers starting in packet 295.

[image]

I don't know if there might be some magic setup or handshaking that occurs previously to the request in 294.

[image]

kyle-github, 2025-04-13:
Thanks very much for this!   Since the values show up in CCW, there must be a way to get them.   Can you make a BOOL array that is global?   I have not used a Micr800 or CCW, so I am not sure what it can do.   I work closely with the author of Pylogix as well and he thinks that the regular method of getting tag data only works on Micro800 for global vars.  Clearly there must be some way to get the others though since CCW shows them!

This will probably take me a while to decode and work through.   

Generally the process that all the other CIP-based PLCs follow seems to roughly be:

- Get the list of tags from instances of some CIP class (it differs between Control/CompactLogix and OMRON).   This is used to get all the tag instance IDs.
- Get attributes about the tags via the class/instance pairs.   Rockwell has a shortcut on some of this to get attributes for all instances via a 0x55 CIP service but that is not supported on OMRON.
- Decode enough of the tag metadata to get the type data.   This tells you whether the tag is an array, a UDT or a regular value like DINT.   If it is a UDT (or an array of UDTs), then ...
- Get all the UDT instance data.   This is usually more difficult than getting the tags.  Not sure why, but, well, Rockwell.

So I will be looking for that kind of pattern in the packet capture.

gotnone, 2025-04-13:
A Few notes:

The BOOL Arrays **are** global

It may be that the local variables are able to be exposed over CIP, but the symbol table remains elusive.

### Missing TAG Attributes

It seems like connecting with the currently running program does not transfer all the information.  There seems to be some streamlining that occurs when CCW has the same program loaded as the Micro850.  It may skip some of the steps in your general outline above when the program on the controller matches with what is loaded in CCW.

In the previous captures my process was:
1. Download the program to the controller
2. Allow CCW to connect online to make sure it was working correctly
3. Disconnect
4. Start the Wireshark capture
5. Go online with CCW
6. Look at the global variable page
7. Stop Wireshark capture

This process left the same program loaded in CCW as what was on the controller.

### Full Featured Capture

I accidentally started a Wireshark capture during the end of the download phase and then i searched for "BOOL" in the packet stream and I saw some interesting results, similar to what you previous expected to see.

The following packet capture uses a different method:
1. Close the CCW project
2. Start the Wireshark capture
3. Use the CCW Discover tool to connect to the Micro850
4. Wait for the project to upload from the controller
5. Look at the global variables
6. Disconnect from the controller
7. Stop the Wireshark capture

This full upload capture also has the interesting "BOOL" text search results

[image]

The packets 251-261 appear to be where the tag details are being transferred.  I see the Global Variables of the BOOL Arrays as well as the IO tags (which are also Global)

#### BOOL Arrays

These appear starting in Packet 251 with the text `[TYPE]` and finish in 253 with what looks like two `\n` characters 0x0a.

#### Global Variables

These appear starting in Packet 253 with the text `[VARIABLE]` and finish in 261 with two `\n`.  The `[END]` test looks like it closes the symbol table response

#### User Data Types?

There is a request for `UDAT` in Packet 262, and a negative response in 263.  I don't have developer edition so I can't make these UDTs.  I may be able to find an example program that uses them if they are still a TODO item for these controllers.

#### Program Upload?

Packets 264 - 479 seem to be passing a large block of data it is an `EZS` request.

#### Local Variable Symbol Table

This is still Missing in Action

Packet 747 and 757 appear to transfer the value `0xFADECAB0` which is one of the magic numbers in the MYBOOLARRAY variable

It is highly likely that this is a red herring and what is being transferred is something like the retentive values.  Some more experiments show that the magic numbers don't show up until I open the local variable page for Prog1.

[image]

[image]

#### Missing Magic Numbers

It appears that the BOOL Arrays do not store their contents in the same order as the other variables.  Searching for these magic numbers doesn't show any significant results besides the red herring above.

### Capture Files

**Use this capture for all packet numbers referenced in this post:**

[CCW_Micro850_Many_BoolArrays_Upload.pcapng.gz](https://github.com/user-attachments/files/19726897/CCW_Micro850_Many_BoolArrays_Upload.pcapng.gz)

The following capture has local variable data transfer beginning when I opened the local variable page for Prog1.  **Do not try to reference the packet numbers from this post with this capture:**

[CCW_Micro850_Many_BoolArrays_Upload_LocalVariables.pcapng.gz](https://github.com/user-attachments/files/19727161/CCW_Micro850_Many_BoolArrays_Upload_LocalVariables.pcapng.gz)

kyle-github, 2025-04-15:
Huh.  Something about the big capture makes my copy of Wireshark crash when I try to drill down.   

There is a _lot_ going on here but most of it looks really weird.  There are several classes and services that look very special and have strange semantics.

Still going through this.