libplctag/libplctag #409: adding libplctag to CMakeLists.txt

ronytigo, 2023-01-05:
I have this CMakeLists.txt file:
```
cmake_minimum_required(VERSION 3.13)  
project(rony_test)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED True)

set(SOURCE_FILES main.cpp)
add_executable(rony_test ${SOURCE_FILES})

#postgres
add_subdirectory(deps/libpqxx build-pqxx)
target_link_libraries(rony_test PRIVATE pqxx)

# influxdb
add_subdirectory(deps/influxdb-cxx)
target_link_libraries(rony_test PRIVATE InfluxData::InfluxDB)
set(INFLUXCXX_WITH_BOOST OFF)

# libplc
add_subdirectory(deps/libplctag)
target_link_libraries(rony_test PRIVATE plctag_dyn)
```

Is this the correct way to add the plctag library (the last two lines in the file)?

I copied the plctag.hpp file and included it in my main.cpp file.
`cmake .. -DCMAKE_BUILD_TYPE=Release` runs successfully but `cmake --build .` fails with:
```
In file included from /home/user/rony/main.cpp:16:
/home/user/rony/./include/plctag.hpp:8:10: fatal error: libplctag.h: No such file or directory
    8 | #include <libplctag.h>
```

kyle-github, 2023-01-05:
I think you need to add a couple of CMake stanzas:

```
include_directories(deps/libplctag)
link_directories(deps/libplctag)
```

You may need to try different directories.   I think the paths are either absolute or relative to the directory in which CMakefiles.txt is found.

I think that the stanza to add the libplctag library for linking may need to be:

```
target_link_libraries(rony_test PRIVATE plctag) # no "_dyn" part
```

Is plctag.hpp your file or did you copy it from somewhere?

I hope this helps!

ronytigo, 2023-01-05:
I copied it from src/wrappers/cpp/include.
Not sure how I should use the c++ wrapper. I'm looking at simple_cpp.cpp and it has `#include "../lib/libplctag.h"`, but I'm not sure if I need to copy the .h file or not.

ronytigo, 2023-01-06:
after building the c library, copying the content of bin_dist to /usr/lib, the simple_cpp.cpp worked.
Thanks!

kyle-github, 2023-01-06:
The C++ wrapper is contributed and has not seen any updated for a long time.  I may need to remove it.   As you found, the C library can be used directly from C++.

kyle-github, 2023-01-08:
Please reopen this or ask on the [forum](https://groups.google.com/g/libplctag) if you have further questions.

closed 2023-01-08 by kyle-github