libplctag/libplctag #410: how to use async_stress.c

ronytigo, 2023-01-08:
Looks like the code creates the same tag:
```
for(i=0; i< num_tags && !done; i++) {
        tags[i]  = plc_tag_create(argv[2], 0);
        statuses[i] = plc_tag_status(tags[i]);

        if(tags[i] < 0) {
            fprintf(stderr,"Error %s: could not create tag %d\n", plc_tag_decode_error(tags[i]), i);
            done = 1;
        }
    }
```
And from the comment in the header: `This example reads a small set of tags repeatedly as fast as possible.`

The first argument is the `num_tags`, but what should I pass as the second one?

kyle-github, 2023-01-08:
This code is used to stress the library.  For that it does not matter what tag or part of tag is used so it defaults to element 0 (zero).  Many of these examples are actually tests of either functionality or performance or correctness.   Since the library is multithreaded and uses asynchronous IO internally and is in C, that is a lot of surface area for bugs!

I should change the comment in the header as that was probably copied from async.c!

What is it that you are trying to do?   

BTW, I am about the only one who will see questions here in GitHub issues.   If you have questions about best practices and generally how to use the library, the [forum](https://groups.google.com/g/libplctag) is the best place.   I might still be the only one who answers, but the responses are easier to find with search and others can see and participate more easily.

kyle-github, 2023-01-10:
Please reopen this or ask on the forum if you still have questions about async_stress.c.   Thanks for using libplctag!