libplctag/libplctag #411: Reading from a VSD

Vadoola, 2023-01-09:
Probably more of a question / discussion than an Issue, but Discussions aren't enabled for this repo.

Is it possible to read parameter vales from something like a PowerFlex 525 VSD? I did a small amount of digging and it looks like the parameters can be read using PCCC encapsulated in CIP (as I understand it, I'm not an expert on the details of these protocols). The Rockwell document for the 525 EIP Adapter (https://literature.rockwellautomation.com/idc/groups/literature/documents/um/520com-um001_-en-e.pdf), talks about dong some explicit messaging to read parameters from the drive.

If I have more time I might dig into this a bit further and see if I can get something working, but I don't have consistent access to a drive, so it might be hard to test.

I thought before I spent a bunch of time on it, I would ask if anyone has considered / tried this before, or knows of some limitation that would prevent it from working.

kyle-github, 2023-01-09:
Probably you should raise this in the [forum](https://groups.google.com/g/libplctag).   The library does support PCCC access to PLC/5, SLC 500 and MicroLogix.  Thanks for the doc pointer.   I do not have any VSD hardware, so this might be tricky to debug.

Vadoola, 2023-01-09:
I completely forgot about the Google Group. I'll post something there. This first came up as an idea weeks ago, I almost forgot I have a Wireshark capture. This is not doing explicit messaging from a PLC (or at least not from PLC logic), but captured doing a 1 second parameter data read from the VSD configuration window within Studio 5000. 

Should someone else want to dig further, I'll attach it here as well as some brief notes, and a CSV of the data that was actually written out from Studio 5000 as a CSV.
[PF525_ParamsRead.zip](https://github.com/libplctag/libplctag/files/10377781/PF525_ParamsRead.zip)

kyle-github, 2023-01-12:
Finally installed Wireshark to read the packet capture.   The order of this seems a little weird.   Was there some of the capture cut off?  It starts off with a set up EIP connection.  Then it uses a slightly weird PCCC command (0x68 which is only supported on PLC/5 or a *Logix emulating a PLC/5).   Older versions of libplctag used that and it is still used when trying to hit a *Logix emulating a PLC/5.   The first packet hits N41:0 and reads some values.  I'll have to look at my code to see what it does here.   

So it looks like if you can figure out what data files it reads, you should be able to set up tags with the PLC type of `logix-pccc` and read/write these files.

Vadoola, 2023-01-13:
So I was trying to do some troubleshooting on a VSD at a customer site when I took that capture. If I recall correctly I started the packet capture, then started capturing some data from the VSD through Studio 5000. At some point I stopped the Packet Capture but left the VSD data capture running, that could be what your seeing about it being cutoff.

I did consider the logix-pccc tag as a way to do this, but wasn't sure how to figure out what tag addresses to use, or how to determine that. I didn't have much more time to look at it, and decided to make this post.

I did do a quick glance around and found it looks like the pycomm3 library can do this, using generic messaging over raw CIP (https://docs.pycomm3.dev/en/latest/examples/generic_messaging_examples.html). I'm not aware of anyway to do that in libplctag, but perhaps I'm wrong. Pycomm3 probably wouldn't work well for my purposes as I was thinking on making a plugin for a software written in C++, not sure how well trying to pull in a python library would work.

I might have a chance to play around with a VSD and see if I can get something working on Monday.

kyle-github, 2023-01-14:
The Wireshark packet dumps show the data files being used.  That is where I got the N41:0 name.  You can do some generic CIP in libplctag using raw CIP packets, but it is definitely cumbersome.

Note that using generic messaging will use CIP class/instance messaging rather than tag names or PCCC-style data files.  It would probably be feasible to build on top of the raw packet interface to make your own library of common commands to the VSDs.   That may be more work than you want to do.

There is an example of using raw CIP packets in the examples in `test_raw_cip.c`.

Vadoola, 2023-01-14:
Yeah, I remember seeing the N41:0 in the packet capture, but how that is associated with the parameters being queried I have no clue. For example I think I was querying b0002, b00037, etc among a few others, which of those was N41:0 and how is that associated. It might be easier to use the raw CIP, as that is how Rockwell has the communication documented so it would probably be easier to determine what you need to query.

I'll take a look at the test_raw_cip.c example, and hopefully I'll find some spare time to try and make something workable.

I'll go ahead and close the issue for now. I'm not sure there is any point leaving it open, and perhaps any more discussion should I find the time to try and flesh this out can take place on the question I created on the forum. Appreciate the help.

closed 2023-01-14 by Vadoola