libplctag/libplctag #412: Is it possible to write to a multi-dimensional array in ControlLogix?

nigeno, 2023-01-10:
I found only one reference to multi-dimensional array here: Unable to read tag from multi-dimensional array #112. It was about reading though and not clear what is the format of the tag path should be in case of two or more dimension.

Let's say the tag path to one dimensional array ArrayINT[1600] is:
protocol=ab_eip&gateway=10.1.100.101&path=1,0&cpu=LGX&elem_size=2&elem_count=1600&name=ArrayINT

How should I change it to write to four dimensional array ArrayINT[1600,4]?

Thank you in advance!

kyle-github, 2023-01-10:
The underlying protocol does not know about multi-dimensional arrays.   You need to multiply all three (there are always three, but some are just "1") together to get the total length:

1 x 1600 x 4 = 6400

```
protocol=ab_eip&gateway=10.1.100.101&path=1,0&cpu=LGX&elem_size=2&elem_count=6400&name=ArrayINT
```

Unfortunately, not all PLCs return all the tag when it is an array.   There are some exceptions.  Omron PLCs always use an `elem_count` of 1 and return the entire tag.

If you have more questions, it is better to ask on the [forum]([url](https://groups.google.com/g/libplctag)) so that others can respond and answers can be found more easily by people searching.

Thanks for using libplctag!

nigeno, 2023-01-10:
Kyle, thank you for your prompt reply.
In this case, the conversion from 1D to 2D array will have to be handled in the PLC, which is definitely possible, but consume extra processing time, or the program in PLC should be changed to operate around 1D, which is not always practical.
If the core of this library allows it would be great to implement n-D array support in the future.
Thank you for your work and support!

kyle-github, 2023-01-10:
I am not sure I understand your statement.  The array is accessed via the network protocol (EtherNet/IP) as if it was one dimension.   If you want to read the entire array, you need to tell the library to request 6400 elements.  

Look at the output of the list_tags_logix.c program for your PLC and you can see how the PLC internally describes the array.   It knows about the dimensions but stores it and retrieves it as if it was a single one-dimensional array.   

Just to make it clear: the network protocol does not have a way to ask for the array in parts.  You get as many elements as you ask for from a given starting point, but the entire array is treated as a one-dimensional array.   If you specify one dimension you have to specify them all.  The following are equivalent:

```
protocol=ab_eip&gateway=10.1.100.101&path=1,0&cpu=LGX&elem_size=2&elem_count=6400&name=ArrayINT
```

And

```
protocol=ab_eip&gateway=10.1.100.101&path=1,0&cpu=LGX&elem_size=2&elem_count=6400&name=ArrayINT[0,0]
```

This is confusing because you _can_ ask for a single element that isn't the starting element, and then get elements from there.  So it looks like multi-dimensional arrays are working.  But they are not.

This is an area where AB/Rockwell exposed their implementation and is not good design.   It is what they did though :-(

kyle-github, 2023-01-10:
I fully agree that this is frustrating!   I made a decision a while ago to not have libplctag internally list all the tags and store their definitions so that the library could make it work a little nicer for users.   The idea was that libplctag (the C DLL) would be a building block for the software that people would use.  Perhaps that was a mistake?   The wrappers (particularly for C#/.Net) have done a nice job of making it easier to use.   But maybe I should create another layer that makes it simpler.

kyle-github, 2023-01-12:
Please reopen this or ask on the [forum](https://groups.google.com/g/libplctag) if you have more questions about multi-dimensional arrays.

closed 2023-01-12 by kyle-github