libplctag/libplctag #413: How to control grouping of tags per request (ControlLogix)

juwalter, 2023-02-10:
Hello @kyle-github - we got pretty far in using your library (many thanks for creating it!) and can successfully list tags and also read them. We are now at the point where we would like to optimize reading, so we can read large groups of tags as fast/as often as possible.

In https://github.com/libplctag/libplctag/issues/107#issuecomment-523156545 you discussed some background for `readgroup` and optimizing reading of tags, specifically from _ControlLogix_, but since this optimization is essentially hidden from the public API, we are wondering how to make use of it. Currently, w are at 220ms for 40 tags but quickly end up at close to 2 seconds for 300 or so tags. which seems quite a lot so I am hoping you could share some advice.

ultimately we would like to read a few thousand tags from the PLC, but of course this will certainly lead to very high latency and possibly put a lot of load on the PLC; therefore, we want to read tags in (small) groups, for example 50 or 100 or something like that.

if we had for example 1000 tags we would like to read, we could group them into 20 groups of 50 tags each. now, should we request them sequentially, or could we perhaps create 20 threads (one thread per tag group) and somewhat read in parallel, or does this completely render the optimizations from `readgroup` moot?

I was wondering about something else, too: my naive thinking would be along the lines: there are 1500 bytes in a ethernet packet, with some luck, this is what the PLC can return in terms of payload; again assuming the response contains all values of the tags from the request, perhaps in order, without and magic bytes as separators then it could be possible to request 1500 tags of size byte from the PLC as maximum optimization.

is this similar to how ControlLogix/CIP works, or is it rather different?

many thanks in advance!

kyle-github, 2023-02-11:
Thanks for using libplctag, @juwalter. 

There is a small write up [here](https://github.com/libplctag/libplctag/wiki/Performance-Tips-with-AB-CIP-based-PLCs) on performance.

The library will automatically try to do two things for performance if you are working with a ControlLogix or a CompactLogix:

1. It will attempt to negotiate a large packet size with the PLC.   Older CIP PLCs generally have roughly 500-byte packet sizes.   Newer ones can support 4000-byte packets.
2. It will attempt to pack as many requests as it can into one packet to the PLC.

For the second item, the library needs you to help it a bit.   Using the asynchronous mode of `plc_tag_read()` and `plc_tag_write()` lets you queue up many requests at the same time.  The library will take as many outstanding requests as it can and pack them into one big request packet to the PLC.

You can also use threading and blocking calls to `plc_tag_read()` and `plc_tag_write()` as long as you use a lot of threads.   Each thread will start a single request to the PLC, but while it waits for it to finish the library can bundle the requests together internally and send them all at once.

Another way to increase the number of concurrent requests is to use the [auto-sync](https://github.com/libplctag/libplctag/wiki/Tag-String-Attributes#optional-generic-auto-sync-attributes) feature and callbacks on completion.   The library will trigger reads (or writes) by itself and will bundle those as it would with the async example.   It will then call your application code when the reads or writes complete (one call per tag).   If you have many tags with different read rates, this might be the easiest way to build your code.

The highest performance that I have measured is when you follow the model in `async.c` and do the following:

- create all your tag handles.
- trigger reads or writes on all of them with a timeout of zero (async mode).
- wait for all to complete by checking the tag status.

I have no problems reading 1000 tags multiple times per second.

There are many factors that can impact performance but the top three are:

- large tags take up more room in response packets.   So fewer responses will come from the PLC.
- long tag names take up more room in the request packets.
- poor network quality can really slow down performance.

I hope this helps.

kyle-github, 2023-03-02:
Please reopen if you have more questions.

closed 2023-03-02 by kyle-github

juwalter, 2023-03-25:
Hello @kyle-github - totally forgot to give feedback (after it worked)

we implemented your suggestion and are seeing very, very good latency (for few tags < 10ms)

Thanks again for your very helpful answer and this fantastic library!!!

kyle-github, 2023-03-25:
Glad to hear that it worked for you!

juwalter, 2023-03-25:
it really did- thanks again!