libplctag/libplctag #414: can't read LINT tags.

ronytigo, 2023-02-11:
This is my code:
```
int64_t val = plc_tag_get_int64(tag, 0);
point.addField("value", val);
```

I get this error:
```cannot convert ‘long int’ to ‘const std::variant<int, long long int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double>&’
  140 |                         point.addField("value", val);
```

I want to get a LINT tag. DINT works fine.

kyle-github, 2023-02-12:
Thanks for using the libplctag library.   Unfortunately, there is not much information to go on here.   The problem appears to be in your C++ code and not in the return value from the library.  The error shows that the type of the return value, int64_t, is not compatible with the type expected by the `addField` method of `point`.  The C++ variant type shows that it can take an `int` (which is why DINT works as that is a 32-bit integer), a `long long int` which might be a 128-bit int, or a C++ string.  None of those match a 64-bit integer.  I suggest you find the implementation of `addField` and add support for 64-bit integers to that.

Please reopen if you have further questions about the library!

closed 2023-02-12 by kyle-github

ronytigo, 2023-02-12:
I used 'long long int = plc_tag_get_int64(tag, 0);`. Is that the correct way?

kyle-github, 2023-02-12:
The definitions of integer types are platform- and compiler-dependent.   Without knowing what compiler you use and what platform you are building for, I cannot say for certain what type you should use.  The error you included shows that the compiler is unable to find a match between `int64_t` (which is apparently `long int` on your compiler+platform) and the `std::variant` type it shows.  Changing the type of `val` to `long long int` might work.