libplctag/libplctag #415: how to read a tag of type REAL

ronytigo, 2023-02-15:
I have a tag of type REAL (controllogix).
Should I use plc_tag_get_float32 or plc_tag_get_float64 in order to read it? Or something else?

timyhac, 2023-02-15:
The Rockwell documentation has good information regarding data types. For the above question, hopefully these will help point you in the right direction:
* Page 23 of the [Logix 5000 Controllers I/O and Tag Data manual](https://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm004_-en-p.pdf).
* Page 12 of the [Logix 5000 Controllers Data Access manual](https://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm020_-en-p.pdf).
* Page 829 of the [Logix 5000 Controllers General Instructions](https://literature.rockwellautomation.com/idc/groups/literature/documents/rm/1756-rm003_-en-p.pdf).

kyle-github, 2023-02-16:
Thanks, @timyhac!

Please reopen this issue if you have any more questions about floating point, @ronytigo.   Thanks for the interest in libplctag.

closed 2023-02-16 by kyle-github

ronytigo, 2023-02-16:
When I convert the values to string, how can I know the precision I should use?
```
            std::stringstream convert;
            convert << std::setprecision(?);
```

timyhac, 2023-02-18:
To answer this question you will need to understand the target audience; who, or what, will read the string representation of that value, and what are their expectations/needs?

Here are some questions that may help:
* Are units important (i.e. millimetres to metres), or does the value need to be mapped before conversion to string (i.e. Celcius to Fahrenheit)?
* Is it intended to be read by a computer program or human? Are there any formatting requirements?
* Is the fidelity of the round-trip conversion important (format binary32 to string, then parse back to binary32) important?

Here are some resources on the floating point formats:
* https://speleotrove.com/decimal/decifaq6.html
* https://floating-point-gui.de/
* https://en.wikipedia.org/wiki/IEEE_754
* https://docs.oracle.com/cd/E19957-01/806-3568/ncg_goldberg.html

ronytesler, 2023-02-18:
@timyhac  I need to pass this float to influxdb, as a string. So I don't want to lose any data, there are no special formatting requirements and no mapping is needed. Just need to pass the value as it as.
Can I just use a large number like setprecision(40) and this way I make sure the number won't be truncated?

timyhac, 2023-02-23:
@ronytesler - there should be an answer to that question in my response.

Note that this question is unrelated to libplctag, it is just general programming/floating-point - if the resources that I've already pointed out aren't enough I trust that you're able to do your own research.