libplctag/libplctag #416: can a tag be in pending status forever?

ronytigo, 2023-02-19:
I call `plc_tag_create(tag_full_path, 0);`
Can the tag be in pending state forever?

timyhac, 2023-02-23:
In context of some of the other questions you've posted, I think what you're asking is "If I call `plc_tag_create(tag_full_path, 0);` against a device that is unavailable (e.g. disconnected/offline), will the status be pending indefinitely?". The answer is yes.

To determine the answer to this question I ran the following C# program:

```csharp
using libplctag.NativeImport;

namespace TestPendingStatusForever
{
    internal class Program
    {
        static void Main(string[] args)
        {
            var tag = plctag.plc_tag_create("protocol=ab_eip&gateway=10.10.10.10&path=1,0&cpu=LGX&elem_count=10&name=MY_TAG", 0);

            while (true)
            {
                Console.WriteLine($"{DateTime.Now} {plctag.plc_tag_status(tag)}"); // shows that the status is Pending
                Thread.Sleep(1000);
            }
        }
    }
}
```

kyle-github, 2023-03-01:
Tim was kind enough to answer this.   Please reopen if you have additional questions.

closed 2023-03-01 by kyle-github