libplctag/libplctag #417: plc not reconnecting

ronytigo, 2023-02-20:
```
plc_tag_create(tag); // this works
// here I disconnect the plc
while (true){
status = plc_tag_read(tag, 0);
sleep(1);
}
```
the status is PLCTAG_ERR_NOT_FOUND (-19) after I disconnect the plc. after 15 seconds I bring the connection up again, but still I get PLCTAG_ERR_NOT_FOUND (-19). I expected to get PLCTAG_STATUS_PENDING and then PLCTAG_STATUS_OK.

timyhac, 2023-02-23:
Is your intention to start a new read every 1ms, and before the previous read finishes?
The async.c example is different to the snippet you've pasted:
https://github.com/libplctag/libplctag/blob/release/src/examples/async.c#L101

kyle-github, 2023-02-26:
Note that you are not passing all the arguments to `plc_tag_create()` or using the return value.  There are two arguments needed.   The first is a string with attributes that defines the tag.   The second is the number of milliseconds to wait for the tag to connect to the PLC and do a first read.   If the number of milliseconds is zero, then the function will return immediately and you will need to check the status of the tag to determine whether it is finished connecting or not.  

The function `plc_tag_create()` returns a 32-bit signed integer.  If the value is greater than zero, then it is a tag handle you use in other functions to reference the tag.   If the value is less than zero, then you have an error and can use `plc_tag_decode_error()` to get a string back.  Your code ignores the return value of `plc_tag_create()` and then uses `tag` later.   Whatever value is in `tag` will be used as a possible tag handle.   That value is highly unlikely to be the actual tag handle, so all subsequent calls into the library will return an error about not finding the tag handle.

Your C compiler should be giving you a warning about missing an argument.   You will be getting a value from the stack or from a register than has not been explicitly set so it could be any value.  Are you new to C coding?

Based on this and your other questions I think you should examine the examples more closely as it seems that you do not really understand how the API works.   Your code above is not going to work for the reason I state above and due to the use of the async read followed by a sleep statement.   If you wish to read a value, then reading with a zero wait (async mode) followed by a sleep will not guarantee that the read completed.  

On Windows, `Sleep(x)`, will sleep x milliseconds.   On a POSIX system, `sleep(x)`, will sleep x seconds.   Notice the case difference.

kyle-github, 2023-03-01:
If you want to see reconnection, you will need to ensure that you have the correct tag string and then correct your code:

```c
int32_t tag = plc_tag_create("protocol=ab_eip&gateway=...", 5000);
while (true) {
    int status = plc_tag_read(tag, 1000); // wait up to 1 second then time out
    printf("status=%d (%s)\n", status, plc_tag_decode_error(status);
}
```

You will need to fill in the "..." part with your tag string attributes.

Once you unplug the PLC, you should see

```
status=-32 (PLCTAG_ERR_TIMEOUT)
status=-32 (PLCTAG_ERR_TIMEOUT)
status=-32 (PLCTAG_ERR_TIMEOUT)
status=-32 (PLCTAG_ERR_TIMEOUT)
status=-32 (PLCTAG_ERR_TIMEOUT)
...
```

After you plug it back in, you should see

```
status=0 (PLCTAG_STATUS_OK)
status=0 (PLCTAG_STATUS_OK)
status=0 (PLCTAG_STATUS_OK)
status=0 (PLCTAG_STATUS_OK)
status=0 (PLCTAG_STATUS_OK)
...
```

Please reopen if you still have questions.

closed 2023-03-01 by kyle-github