libplctag/libplctag #418: Can not write values to tag, either erroroutof bounds or errorbadconfig

kaptindan, 2023-02-28:
I am scratching my head. I can not write to a plc tag. Allen Bradley control logics. I have a string and the plc type is string. I've tried many different data types. 

	var myTag13 = new Tag<StringPlcMapper, string>()
			{
				Name = "Scanner_P502_3A_WCS.CMD",
				//Name = "Program:P502.Scanner_WCS.TRIG",
				Gateway = "192.168.11.1",
				Path = "1,0",
				PlcType = PlcType.ControlLogix,
				Protocol = Protocol.ab_eip,
				Timeout = TimeSpan.FromSeconds(5)
			};
			myTag13.Initialize();
			myTag13.Value = txtScannerWCSTrig.Text;
			myTag13.Write();

The string in txtScannerWCSTrig is 31200081420000437705

PLC tag is a STRING_50

timyhac, 2023-02-28:
https://github.com/libplctag/libplctag/wiki/Strings-and-String-Handling

Use non generic `Tag` instead of `Tag<M,T>` to access the string configuration parameters.

kyle-github, 2023-03-01:
From the type name, I assume that this is not a standard STRING type but something that is defined in the program.   In that case, please do as @timyhac mentioned.   This looks to be specific to the .Net wrapper rather than the core C DLL.   Please direct questions related to .Net to the issue you opened in that repo.

Reopen this issue if you have questions about the core C DLL.

closed 2023-03-01 by kyle-github

kaptindan, 2023-03-01:
Hey thanks for the response! I am very new to talking to a PLC. I've done it in the past using libplctag but only changing bits and writing to a basic string. So this is a PLC is one that someone has already programmed using string_50 like below. I read over the link strings and string handling and still not sure how to over come this. lol

kaptindan, 2023-03-01:
Ok thanks I will repost in the libplctag.NET post.