libplctag/libplctag #420: Memory Leak on Specific PCs

jewell9, 2023-03-20:
Hi All,

I created a winforms application using libplctag that is causing a memory leak ONLY on shop floor PCs. 

I created a component that scans a PLC tag every second to update my winform. I have published this app (64bit, release) and ran it on both my laptop and a thinclient. While it runs just fine on both systems, the thinclient gradually uses more and more memory as time passes, while my laptop does not increase whatsoever.

I have already ruled that the issue is relating to the libplctag library as this does not happen when I remove all functions utilizing the library.

I downloaded a profiler and found that **plctag.DLL!plc_tag_write+0x1630** is using way more resources on the thinclient compare to my laptop.

I am disposing on every call.

What I believe I have narrowed it down to - are there any minimum hardware or system requirements for this library to run without issues? Any help is greatly appreciated.

[image]

Thanks

kyle-github, 2023-03-21:
@jkoplo ?   @timyhac ?   Can you help here?

timyhac, 2023-03-21:
An example of logging is [here](https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/LoggingExample.cs).

My reading of the above is that "thinclient" refers to the usage of the device, as opposed to anything about the device itself, or a build of Windows - is this correct?

There are no minimum hardware/system requirements for libplctag,  nor libplctag.NET (at least in the context of what is being discussed here).

.NET Runtimes do change their behaviour depending on the environment they're running in. E.g. environments with lower memory, the garbage collector will be more aggressive.

Is it intentional to dispose on every call. From the manual:

> Tag handles are designed to be very long lived. They can live weeks or months. In order to use the library to its full potential, it is important to note this...

timyhac, 2023-03-21:
Btw - normally for IDisposable objects it is better to use the `using` statement so that it can dispose of objects even if you get exceptions. In the code you posted, what happens if Read(); throws an exception?

```csharp
public static bool GetBoolTag_ControlLogix(string ip, string path, string name)
{
    try
    {
        var boolTag = new Tag<BoolPlcMapper, bool>()
        {
            Name = name,
            Gateway = ip,
            Path = path,
            PlcType = PlcType.Omron,
            Protocol = Protocol.ab_eip,
            Timeout = TimeSpan.FromSeconds(5)
        };
        
        boolTag.Initialize();
        bool readValue = boolTag.Read();    // If this throws exception
        boolTag.Dispose();                  // Dispose will never run
        return readValue;
        
    }
    catch (Exception ex)
    {
        return false;
    }
}
```

I trust this code was just used for debugging - it can't differentiate between whether the Tag had a false value, or if there was an exception.

Also note that DIspose is only disposing of native resources, not of .NET memory - the Garbage Collector will do that when it thinks it is needed. You can force it to run however, maybe this is a line of questioning to investigate.

jewell9, 2023-03-21:
Good catch. I will give that a shot. I have a feeling it could even be network related with the shopfloor machine. Maybe it isn't hitting those PLCs and that could be exactly what is happening. I will look into adding the using statement. Do you have any documentation with this already in place?

Correct. The returning of false as the exception is simply for debugging at the moment.

timyhac, 2023-03-21:
No we don't have any examples of this - the Tags are intended to be long lived so you should be keeping the Tag objects in memory somewhere, and make repeated calls to Read on the same Tag object.

closed 2023-03-22 by jewell9