libplctag/libplctag #422: Performance tests returning unexpected results

fabio160781, 2023-04-10:
Dear All,
My team has been running some tests using the libplctag.net library and some of our tests had raised questions we are not being able to answer. First of all, we highly appreciate the interactions we had in the past with you all and we are very interested in this library. One of my colleagues have also been talking to the .net team but this time I believe the "issues" had to be created in the libplctag (more related to the c++ library). 

Our tests basically consists on performance tests. We want to make sure the library is using all the resources to get the best performance possible (which is extremelly important to our software) that provides user interface for our customers to configure thousands of address/tag communication. The users can create different spreadsheets (each spreadsheet creates a new connection).

We have been testing the library using the following PLC.
**Plc Info**
Type: 1769-L35E CompactLogix5335E Controller
Revision: 20.19
System Overhead Time Slice: 45%

1) During our tests, we used wireshark to monitor the read operations with multiple tags and we have noticed that when setting "AllowPacking" does not group multiple tags into the same read request.

[image]

The image above shows there is a request per tag (Service 0x52). 
From another thread, the comment below was added:
`This packing is transparent to the application but you need to help the library by queuing the requests at the same time.  `

Apparently we are not queuing the requests fast enough? Is there an alternative to queue requests and make sure they will be grouped into the same package? Or do you see any other reason why we can't use the packing feature?

2) Our next test was enabling the auto read (using the auto sync configuration), this time we could see the multiple packing after monitoring wireshark, but we get to a second doubt. 

According to the PLC responses, our PLC does not support Large Forward Open (bigger packages) as you can see in the image below

[image]

And according to Large_Forward_Open and Forward_Open definition below

[image]

Even though we won't be able to send large packages, we could still be sending up to 500 bytes, but after monitoring the communication with auto read, we don't see our packages reaching the 500 bytes.

The image below shows the communication with different package sizes

[image]

Do you know if the library really uses full capacity to create the requests? Why we are not being able to use the 500 bytes capacity in all the requests? Does it uses full capacity in your tests?

Thanks in advance, I hope I was able to explain properly our tests. If you need more information please let me know.

lcoeli, 2023-04-15:
@kyle-github any thoughts?

timyhac, 2023-04-16:
@fabio160781 - are you able to try concurrent reading of multiple tags using the async read instead of auto-sync read?

fabio160781, 2023-04-17:
Dear Tim,
**I believe it can be done, but does the allow packing only work for async read?** 

Since our software is similar to an IDE, the users have control on how they want to configure the applications they are building which also includes the communication they are creating using the libplctag. So it is important for us to understand the auto-sync read and the allow packing because this will be a user configuration and needs to be well documented.

kyle-github, 2023-04-17:
The library is internally multithreaded and will pack as many requests as it has room for and are queued up for transmission to the PLC.  There are multiple ways to help the library do that:

1. use multiple threads with one tag per thread.  All library calls can be synchronous, but you need a thread per tag.
2. use async reads and writes.
3. use auto-sync.

There are a few phases that the library goes through to set up a tag:

1. Open a TCP socket to the PLC.
2. Handshake to set up an EIP session.
3. Handshake to set up a CIP connection.
4. Read the value of the tag from the PLC.

Only after all these parts are done is the tag ready for user application use.   If you check the tag status before these are all done, you will get PLCTAG_STATUS_PENDING as the status.

If you use synchronous tag creation, then the library will do all of these steps before returning to your application.

Does that help?

timyhac, 2023-04-17:
I'd imagine that if you were debugging, and you wanted to force packing to see whether it was working, it would be useful to be able to control the timing of reads/writes to provide libplctag the best chance of being able to perform packing.

Since there is no control over precise timing for auto-read, it would not be suitable for debugging.
For example, if you had 2 tags, both set to read every 1 second, - would they both read at [0, 1, 2, 3] - or would one read at [0,1,2,3], and the other at [0.5, 1.5, 2.5, 3.5]?

On the other hand, the other two options would be suitable for debugging:
1. Multi-threading - you could use threading primitives to ensure that the reads are triggered at the same time.
2. Async - you could trigger the two reads at the same time.

kyle-github, 2023-04-17:
On the above, note that the first three steps of the tag set up process are only done once per PLC.  The fourth part, reading the initial tag value, is done for every tag you open on that PLC.

fabio160781, 2023-04-19:
Thanks for the suggestions @kyle-github and @timyhac. Your explanation does help us understand what happens internally in the lib. 
We will implement async reads and writes and monitor the requests using wireshark. I will keep you posted.

kyle-github, 2023-04-22:
Due to timing, it is possible that the library will not be able to pack the maximum amount of requests into a single packet to the PLC.   If you have many, many tags, then it will definitely be able to pack the maximum number of requests into a single packet to the PLC.  If you have about the number of tags as will fit into a single packet, then there is a meta-stable state you can get into where some of the requests go into one packet and the rest go into the next packet.

fabio160781, 2023-04-26:
We were able to test async communication (read) and the pack worked as suggested by you. There were significant performance improvements during our tests but we also observed a few errors that at the time being, we still didn't identify if its root cause is on our code or inside the libplctag library (some other priorities came up but we will be investigating them as soon as possible). I will post more comments to this thread as soon as we have more information.
Thanks again for your comments @kyle-github and @timyhac

kyle-github, 2023-04-27:
Thanks for the update.   Please let me know if you are finding errors!   Error bugs are the most important thing to fix for libplctag!

timyhac, 2023-07-04:
Hi @fabio160781 - I note that it has been about 3 weeks since [issue #342](https://github.com/libplctag/libplctag.NET/issues/342) has been resolved and I believe this is the problem you were referring to above. Ok to close out this ticket?

fabio160781, 2023-07-04:
Dear @timyhac, you are correct, issue #342 was directly related to this ticket and we were able to fix it. We can close this ticket.
Thank you for the support !

kyle-github, 2023-07-21:
Closing.  Thanks for bringing this up.

closed 2023-07-21 by kyle-github