libplctag/libplctag #425: which threads run the callbacks?

ronytesler, 2023-04-12:
I use `plc_tag_create_ex` and `plc_tag_register_callback` from different threads, and using different callback methods (as the tag_callback_func/tag_callback).

On which thread or threads will the callbacks run?

kyle-github, 2023-04-16:
It depends where the event is emitted and how the library is being used.   Some events, like read start, are emitted in the main API code and thus run in your application's calling thread.   Some events like read complete can come from more than one locations depending on how the library is called.   However, they will be called from either the IO thread for the PLC or from a global tag tickler thread that handles things like auto-refresh tags.

kyle-github, 2023-06-02:
Please reopen if you have any more questions!

closed 2023-06-02 by kyle-github