libplctag/libplctag #426: How to read Access Level of Tags and Usage of UDTs?

JulZimmermann, 2023-06-29:
Hello,

is it possible to read the Access Level (Read, Write, Read/Write, etc.) of Variables and Usage of UDTs? (Input, Output, InOut, Local, etc.) of UDTs?

Tags in Logix Designer:
[image]

UDTs in Logix Designer:
[image]

Thanks!

timyhac, 2023-07-04:
This is not exposed in libplctag.

Quickly browsing the Rockwell documentation there was some info on this:
* [pg12 of the Logic 5000 Controllers Data Access manual](https://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm020_-en-p.pdf)
* [page 60 of the Logix 5000 Controllers I/O and Tag Data manual](https://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm004_-en-p.pdf)

JulZimmermann, 2023-07-04:
Hi.

I looked at [pages 39 and 48 of the documentation ](https://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm020_-en-p.pdf)

If I understand correctly, the information I wanted to access is not available via the `Get_Instance_Attribute_List (0x55)` and `Get_Attribute_List (0x03)` services.

Thank you anyway!

joed74, 2023-07-05:
Maybe it is accessible via `Get_Instance_Attribute_List (0x55)` and Attribute 10:

[image]

Source: https://docs.pycomm3.dev/en/latest/api_reference/logix_driver.html#pycomm3.LogixDriver.get_tag_list (scroll down to function get_instance_attribute_list_service)

kyle-github, 2023-07-21:
Thanks for the tip.   This looks interesting.  I wonder if just always requesting attribute 10 will work?   Each attribute has a separate status response, so it might work as long as we can see that asking for attribute 10 fails on some systems and the others succeed.

I wonder why the pycomm3 author is pulling attribute 6?  I found attribute 8 by experimentation.

tomatac, 2023-09-13:
It would be really nice to be able to read tags that are based on UDTs.
I am curious, what is the currently the best practice on that?