libplctag/libplctag #427: omron ethernet/ip writing error

lupokk, 2023-06-30:
I think there is a problem writing an unsigned byte to omron device with ethernet/ip

timyhac, 2023-06-30:
Hi @lupokk - welcome to the libplctag community.

Thanks for taking the time to contribute to the project and we would love to know more about the problem you're facing.

Are you able to provide some additional details? On the libplctag.NET subproject we have some guidance on additional information you can provide: https://github.com/libplctag/libplctag.NET/blob/master/CONTRIBUTIONS.md#guidance-when-creating-an-issue

lupokk, 2023-07-04:
With the example below I get an error "LibPlcTagException: ErrorRemoteErr"

 var myTag = new Tag<SintPlcMapper, sbyte>()
         {
            Name = "EIP_TO_COGNEX2.RESERVEDBYTE_0",
            Gateway = "172.17.252.11",
            Path = "1,0",
            PlcType = PlcType.Omron,
            Protocol = Protocol.ab_eip,
            Timeout = TimeSpan.FromMilliseconds(2000),
         };

         await myTag.InitializeAsync();
         
         myTag.Value = Convert.ToSByte(37);

         await myTag.WriteAsync();

kyle-github, 2023-07-21:
This is probably more a question for the libplctag.NET project as Tim noted above.   That said, can you capture a log file?   Please set the debug level to 4 and capture the log output.   That will hopefully tell us more about what is happening.

kyle-github, 2025-03-04:
Please reopen if you still have problems.  Thanks!

closed 2025-03-04 by kyle-github