libplctag/libplctag #428: Add support for FINS?

oPiZiL, 2023-06-30:
> The FINS/UDP protocol is an Omron protocol which can be used by a PLC program to transfer data and perform other services with a remote PLC connected on an Ethernet Network. The FINS protocol allows Omron PLCs to communicate with the [RMC75E](https://deltamotion.com/support/webhelp/rmctools/System_Components/RMC70/CPU_Modules/RMC75E_Module.htm), [RMC150E](https://deltamotion.com/support/webhelp/rmctools/System_Components/RMC150/RMC150E_Module.htm), and [RMC200](https://deltamotion.com/support/webhelp/rmctools/System_Components/RMC200/RMC200.htm). The RMCs support FINS/UDP.  

https://github.com/lammertb/libfins

kyle-github, 2023-07-21:
I raised this question with some people that know people at Omron and was told that FINS is on its way out and most Omron customers use CIP.  Thanks for pointing out the library you have above!   That looks very interesting and has a usable license.

kyle-github, 2023-07-21:
Omron's implementation of CIP is not great.   It does not match Rockwell's very closely.

kyle-github, 2024-06-09:
Marking this as WONTFIX and closing.  There are other libraries that support Omron's version of CIP and other users are contributing code to support that.

closed 2024-06-09 by kyle-github