libplctag/libplctag #429: Write MyTag[1] error

eslinan, 2023-07-02:
I start ab_server, and use pylogix to write tag like this is Ok:
from pylogix import PLC
with PLC() as comm:
    comm.IPAddress = '127.0.0.1'
    value = [1, 2, 3, 4, 5]
    comm.Write('MyTag', value)

but when I write to MyTag[1], it does nothing, why? Thank you.

timyhac, 2023-07-04:
Hi @eslinan - welcome to the community.

pylogix is obviously a [separate project](https://github.com/dmroeder/pylogix), but assuming you're asking about ab_server - you might need to provide some more information about what you're expecting to happen, and what is actually happening.

On the sister project libplctag.NET we have some guidance on how to create an issue - see https://github.com/libplctag/libplctag.NET/blob/master/CONTRIBUTIONS.md#guidance-when-creating-an-issue

eslinan, 2023-07-05:
I want to write a list to tag begin with index 4, not index 0.

kyle-github, 2023-07-21:
As Tim noted, this is for a different project than libplctag.  Please open a ticket there or join the pylogix Discord server.  Dustin (the Pylogic author) is a great guy and more than happy to help with all kinds of questions.

closed 2023-07-21 by kyle-github