libplctag/libplctag #432: Cannot connect to EN2T help

nigeno, 2023-07-25:
Cannot connect to plc thru ethernet/ip bridge - comms module EN2T.

Processor module located in slot1 with ip address of 191.168.158.39, EN2T in slot2 with ip address of 192.168.1.10.

According to the issue  Handling Embedded Gateways #297 instructions, the path should be:
"protocol=ab_eip&gateway=192.168.1.10&path=1,2,18,191.168.158.39,1,1&cpu=lgx&&elem_size=2&elem_count=8&name=TestTag"

Comms doesn't work though. Please help!

Couldn't find any API instruction in this regard except the issue comments, is this feature even working now?

kyle-github, 2023-07-25:
Hi.  Thanks for using the library.

I am trying to understand how your PLC is set up.   Based one what you've said, you have the following slots in a ControlLogix chassis:

slot 0: ?
slot 1: CPU module (L80 or later?), IP address 191.168.158.39 (did you mean 19**2**.168.158.39?)
slot 2: EN2T, IP address 192.168.1.10
slot 3: ?
...

Normally CPU modules are placed in slot 0, so you may need to check that.   The first slot is slot 0 (zero), not slot 1 (one).

From your description, there is no need to use bridging.   

How are you getting to the PLC?   On the same network as the CPU module?  If so, then your tag string would be "protocol=ab_eip&gateway=191.168.158.39&path=1,1&cpu=lgx&elem_size=2&elem_count=8&name=TestTag".

If you want to talk to the PLC from the network attached to the EN2T, then the connection string would be "protocol=ab_eip&gateway=192.168.1.10&path=1,1&cpu=lgx&elem_size=2&elem_count=8&name=TestTag".

Please double check that you really meant 19_1_.168... for the CPU IP address.  Are you sure it is not 192.168...?

The gateway is the IP address that is first encountered when trying to access the PLC.  It is the next "hop" in the network from the computer that is running libplctag.   The path indicates how to route the requests to a CPU module.   A path of "1,1" means "go to the backplane" (which is port 1 out of the network module), then "go to slot 1" in the back plane.

If you actually have the CPU in slot 0, then you need to change the slot numbers above to zero (0).  The paths would be "1,0".

Hopefully this helps.

nigeno, 2023-07-25:
Thank you for a prompt reply!

Slot 0: empty
Slot 1: L83E, and yes, it has a weird ip 191.168.158.39
Slot 2: EN2T, 192.168.1.10
Slot 3: Analog card
...
I'm getting to plc thru the remote IO network, that is, 192.168.1.XXX subnet.

But, why you mentioned "protocol=ab_eip&gateway=192.168.1.10&path=**1,1**&cpu=lgx&elem_size=2&elem_count=8&name=TestTag", if the EN2T module located in slot2, then it should be 1,2? I will try this now

nigeno, 2023-07-25:
While waiting for a chance to test it (cannot do it right now), just to clarify:
if you connect to plc thru EN2T module, you just put one ip address? not two like I mentioned in my original question? Probably it was a confusion on my side with the issue #297.

Again if I connect using EN2T, it's in slot 2, then the total path would be like follows?
"protocol=ab_eip&gateway=192.168.1.10&path=1,1&cpu=lgx&elem_size=2&elem_count=8&name=TestTag"

So It's regardless of cpu module slot # and its ip address?

nigeno, 2023-07-25:
It worked!
With the following path:
"protocol=ab_eip&gateway=192.168.1.10&path=1,1&cpu=lgx&elem_size=2&elem_count=8&name=TestTag"

So, as I understood you have to put an IP address of the gateway you are talking to, regardless whether it's cpu module or a comms module, then you put a path including first "1" for the backplane, and second "slot ##" of where the cpu module is installed.

Thank you for your library!

closed 2023-07-25 by nigeno

kyle-github, 2023-07-25:
Glad to hear it worked.  The path is "forward only" trail that points to where you want to go, not where you were.  So the path "1,1" means:

- go to the backplane of the chassis (that is port 1 on a module like DHRIO or ENBT or EN2T).
- go to slot 1 from the backplane.  Slot 1 is where your CPU module is.

The path tells the request where to go not where it was.   The IP address is on the EN2T.

kyle-github, 2023-07-25:
Also, note that the element size is optional if you are using a ControlLogix or most other CIP-based PLCs.