libplctag/libplctag #433: Reading AB PLC Fault State

bcurless1, 2023-08-15:
Hi,

Is there any way to read the PLC fault state on a CompactLogix plc that you know of? I've previously queried the CPU state using @CPU_STATE in ingear - [Ingear Documentation](https://ehelp.ingeardrivers.com/dotnet/core/logix/WebHelp/netlogix.htm). It ends up being very helpful because if the PLC has a major fault then the program won't run and the fault can't really be exposed by means of other publicly available tags being updated. I attempted to read the tag "@CPU_STATE to no avail, but figured I'd ask before digging too deep into this.

Thanks!

kyle-github, 2023-08-16:
There is, but it is a bit involved.   It is not directly supported in the library.  You need to use the raw tag approach.  I will be coding up an example in the future that will be included with the library.   There is an ongoing discussion on how to decode the results in the pylogix Discord that is very detailed.   Jeremy Medders is doing most of the research.   He will be adding onto his wiki [here](https://github.com/JeremyMedders/LogixLibraries/wiki) when he has more information.

bcurless1, 2023-08-16:
Oh wow very interesting, thanks for the link to the discord. Thanks for all this info! I'll check back in the future.

closed 2023-08-16 by bcurless1