libplctag/libplctag #434: writing problem in omron nx1p2 PLC

tly-seven, 2023-08-29:
Recently when using libplctag to connect to Omron nx1P2 PLC, I found a problem, when writing data to a byte-sized tag, it will report an error, but there is no problem to read the data, and there is no problem for other data types, see the attachment for the code and log file. Can anyone help take a look at this? Some help would be greatly appreciated!
@kyle-github @timmot @scontini76 @Joylei @alimon 
======================================================================================
code:
#include <unistd.h>
#include <signal.h>
#include <stdarg.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include "libplctag.h"

#define REQUIRED_VERSION 2,4,0
#define TAG_PATH_0 "protocol=ab_eip&gateway=192.168.250.1:44818&path=18,192.168.250.1&plc=omron-njnx&name=usint_t"
#define DATA_TIMEOUT 5000

int main(){

    plc_tag_set_debug_level(2);

    int32_t tag = 0;
    char tag_string[256]={0};
    char error_string[128]={0};
    int rc = PLCTAG_STATUS_OK;
    int tag_size = 0;
    int e_size = 0;
    int e_count = 0;
    uint8_t ui8 =0;

    /* create the tag. */
    tag = plc_tag_create(TAG_PATH_0, DATA_TIMEOUT);

    /* get the data */
    rc = plc_tag_read(tag, DATA_TIMEOUT);

    tag_size = plc_tag_get_int_attribute(tag, "size", 0);
    e_size = plc_tag_get_int_attribute(tag, "elem_size", 0);
    e_count = plc_tag_get_int_attribute(tag, "elem_count", 0);
    ui8 = plc_tag_get_uint8(tag,0);
    printf("elem_count : %d\n",e_count);
    printf("elem_size : %d\n",e_size);
    printf("tag_size : %d\n",tag_size);
    printf("%u\n",ui8);
    printf("0x%x\n",ui8);
    printf(" ========================================= \n");

    plc_tag_set_uint8(tag, 0, 7);

    tag_size = plc_tag_get_int_attribute(tag, "size", 0);
    e_size = plc_tag_get_int_attribute(tag, "elem_size", 0);
    e_count = plc_tag_get_int_attribute(tag, "elem_count", 0);
    ui8 = plc_tag_get_uint8(tag,0);
    printf("elem_count : %d\n",e_count);
    printf("elem_size : %d\n",e_size);
    printf("tag_size : %d\n",tag_size);
    printf("%u\n",ui8);
    printf("0x%x\n",ui8);

    rc = plc_tag_write(tag, DATA_TIMEOUT);
    if(rc != PLCTAG_STATUS_OK) {
        sprintf(error_string,"ERROR:= Got error code %d: %s\n",rc, plc_tag_decode_error(rc));
        printf("%s\n",error_string);
        plc_tag_destroy(tag);
    }

}
==================================================================================
log
2023-08-29 14:43:16.796 thread(1) tag(0) INFO plc_tag_create_ex:838 Starting
[... 603 lines cut ...]
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL mutex_destroy:774 Done.
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL mb_teardown:2456 Modbus mutex destroyed.
2023-08-29 14:43:16.838 thread(1) tag(0) INFO mb_teardown:2458 Done.
2023-08-29 14:43:16.838 thread(1) tag(0) INFO lib_teardown:176 Tearing down library.
2023-08-29 14:43:16.838 thread(1) tag(0) INFO lib_teardown:181 Signaling tag tickler condition var.
2023-08-29 14:43:16.838 thread(1) tag(0) INFO lib_teardown:186 Tearing down tag tickler thread.
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL thread_join:872 Starting.
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL thread_join:884 Done.
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL thread_destroy:913 Starting.
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL thread_destroy:924 Done.
2023-08-29 14:43:16.838 thread(1) tag(0) INFO lib_teardown:193 Tearing down tag tickler condition var.
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL cond_destroy:1182 Starting.
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL cond_destroy:1196 Done.
2023-08-29 14:43:16.838 thread(1) tag(0) INFO lib_teardown:199 Tearing down tag lookup mutex.
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL mutex_destroy:758 Starting to destroy mutex 0x7f779b60e8c0.
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL mutex_destroy:774 Done.
2023-08-29 14:43:16.838 thread(1) tag(0) INFO lib_teardown:205 Destroying tag hashtable.
2023-08-29 14:43:16.838 thread(1) tag(0) INFO hashtable_destroy:260 Starting
2023-08-29 14:43:16.838 thread(1) tag(0) INFO hashtable_destroy:272 Done
2023-08-29 14:43:16.838 thread(1) tag(0) INFO lib_teardown:212 Done.
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL mutex_destroy:758 Starting to destroy mutex 0x7f779b60e890.
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL mutex_destroy:774 Done.
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL plc_tag_unregister_logger:1368 Starting
2023-08-29 14:43:16.838 thread(1) tag(0) DETAIL plc_tag_unregister_logger:1372 Done.
2023-08-29 14:43:16.838 thread(1) tag(0) INFO plc_tag_shutdown:1142 Done.

kyle-github, 2023-10-14:
There is a CIP error, 0x1f, that is being returned from the PLC.   Unfortunately, I can't find a definite source of what that means.   The most I have been able to determine is that it is "vendor specific".

However, I do a possible issue here.   It looks like the library is writing two bytes of data to the PLC instead of one.   The packet is correct but there is a trailing zero byte that looks like it should not be there.   I can read and write to Rockwell PLCs so it is possible that this is some sort of Omron-specific problem.

I am going to call this a bug and see if I can get rid of that extra byte.

kyle-github, 2024-06-07:
This does look like a bug.  However there is code there specifically to pad everything out to 16-bit boundaries.   I am going to comment that out and run some tests to see what happens with my other PLCs.

kyle-github, 2025-03-04:
I lost track.  Does release 2.6.4 Beta fix this for you?  If not, please reopen!

closed 2025-03-04 by kyle-github