libplctag/libplctag #436: Unknown data type DT

joed74, 2023-09-15:
I cannot read DateTime (DT) fields in my CLX plc:

2023-09-15 16:27:27.822 thread(3) tag(0) INFO unpack_response:1651 00048 00 00 **c0** 00 63 94 90 97 66 05 06 00
2023-09-15 16:27:27.823 thread(2) tag(0) WARN check_read_status_connected:1448 Unsupported data type returned, type byte=**192**

The underlying datatype is just an ULint (or Lint)

I think it has to be added to defs.h

#define AB_CIP_DATA_DT         ((uint8_t)0xC0) /* DT value, 64 bit */

[image]

Interestingly, the LDT data type (called STIME in libplctag, type 0xCC) works out of the box

kyle-github, 2023-10-11:
Unfortunately, the types are not really used anywhere.   Only atomic values seem to be directly supported.

kyle-github, 2023-10-11:
But I can add this in for what it is worth.

kyle-github, 2024-06-09:
This should be in the next release.  I really need to find a good list of these type bytes.

closed 2024-06-09 by kyle-github