libplctag/libplctag #437: Unable to Read tag from Allen-Bradley 1756-L73 ControlLogix PLC

ahmad9918, 2023-09-20:
Hi,

I am trying to communicate and read tag value of Allen-Bradley 1756-L73 ControlLogix PLC which is throwing error as Service not supported: Connection Manager - Large Forward Open . After this attempt is made for Forward Open followed by TCP retransmission packet. I am attaching Wireshark logs for your reference. Please let me what could be the reason for this failure.

Wireshark Screenshot :

[image]

ahmad9918, 2023-10-07:
Hi,

Can you please share your inputs on above mentioned issue.
I am blocked due to this problem.
Thanks

kyle-github, 2023-10-07:
Newer PLCs, and Ethernet modules, support the newer extended Forward Open.  Older ones do not.  This is how the library finds out what it can do when talking with the PLC.  It will try the extended version first and if that fails with an unsupported error, it will fall back to trying the older version.  

I have not seen the TCP retries like that though.  Can you capture log output with log level 4?  I need to see what the library is doing here.   Thanks for bringing this up.

ahmad9918, 2023-10-07:
Hello Kyle,

Thanks for responding.
I have tried to run simple.c from the example, below logs captured with log level 4.
Find find the attached text file for your reference. 
[LogOutputWithDebugLevel4.txt](https://github.com/libplctag/libplctag/files/12838299/LogOutputWithDebugLevel4.txt)

kyle-github, 2023-10-07:
Thanks for the logs.   Here is what I see the library doing:

- it sends the ForwardOpenEx.
- it receives an Unsupported service error from the PLC or network module.
- it sends an old-style ForwardOpen.
- it waits for a response.
- and waits...
- and waits...

What kind of network module are you using for these communications?   Is there any firmware upgrade you can apply?   I have L55, L61 and L81 CPUs with a few different network modules.  The L81 has a built in module.   I have never seen this behavior.

I am going to call this a bug as the library should be able to handle this.  From your Wireshark dump, it looks like the PLC/module drops the TCP connection after sending the Unsupported error.   However, it drops it in a way that almost looks like the network module is crashing.   The TCP connection is not closed by the network module.  It simply disappears.   The packets after that are from the PC/library's TCP stack trying to contact the PLC.

kyle-github, 2023-11-03:
This might take a little longer than I thought.   I thought it would be fairly straightforward to change the state machine here, but it looks like I did not follow good coding practices and have logic smeared across multiple states.

kyle-github, 2024-04-19:
This is still in progress.  I am really getting confused by this.   None of my PLCs do this.  I have a PLC/5, a MicroLogix 1100 and some ControlLogix including ENBT modules and a L81 with built-in Ethernet.

kyle-github, 2024-06-09:
I still do not have a good answer for this.  I have had others do some testing and no one can recreate this problem.   However, I do not know of anyone with your exact combination of hardware and firmware versions.

I think one way to fix this is to add an attribute that suppresses use of the extended ForwardOpen command. You would put that into your tag string.   I will update this when I have an idea what this will look like.

kyle-github, 2024-06-09:
I am removing the critical label as no one else seems to have this issue.  It is till a bug!

kyle-github, 2024-06-09:
I added a new attribute `conn_only_use_old_forward_open`.  If you set it to 1 in your tag string, it will prevent the library from using the new extended ForwardOpen command.

```
protocol=ab-eip&gateway=10.1.2.3&path=1,0&plc=logix&conn_only_use_old_forward_open=1&name=TestBigSINTArray[0]
```

This will be out in the next release.  Probably v2.5.7.

closed 2024-06-09 by kyle-github