libplctag/libplctag #439: Notification when all of the tags have been read with "auto_sync_read_ms"...?

ArjanVer, 2023-11-14:
Hi,

thanks for developing this very nice library!
I am currently developing an application based on libplctag which is able to connect to various PLC types. The user here can select a reading cycle which will be then passed to the "auto_sync_read_ms" parameter.

I am looking for ways to detect when **all** of the configured tags have completed reading for in one cycle. 
For example PCCC PLC models like SLC and MicroLogix only support around 100 tags for 1 second, but theoretically a user could configure reading 1000 tags for 1 second.
This will stretch the reading time, since the PLC will be under load and won't be able to read all of the 1000 tags in the given 1 second. Now I would like to detect this and give a feedback to the user, by tracking the time it takes for all of the 1000 tags to be read for example. The library handles each tag individually, so that it is difficult to get a notification when all of the 1000 tags have been read.

Is there any way I could achieve this with the current implementation?
I thought of tracking the time for the first time without "auto_sync_read_ms" and then setting "auto_sync_read_ms", however this is not very good since the fetching time could change in future...

timyhac, 2023-11-14:
One thing to keep in mind is that not all tags will complete in the same amount of time, so you might need to have different cycle times for different tags (i.e. individual DINT tags could take 50ms per read, but a huge array of 1000 DINTs could take several seconds).

If it is important that each tag is read only once per cycle, then you should instead use the asynchronous read (i.e. where timeout=0). You can use your own periodic timer to start the tag reads once per cycle, and perform timeout checks either on a per-tag basis or for the entire set of tags.

Many other things can go wrong besides the read taking longer than expected, so you may consider performing additional logic to check for other failure modes.

kyle-github, 2023-11-16:
The idea of a cycle is not really something that the library understands.  Each tag is treated independently.  Internally there are some shared things (like the connection to the PLC).   

As @timyhac says, if you want to have your own concept of a cycle, you need to impose it by using async mode.

kyle-github, 2023-11-19:
Was any of this any help?

Unfortunately, PLCs like Micro8xx, PLC/5, SLC 500 and MicroLogix do not support multiple requests per packet to the PLC.   Only Control/CompactLogix (and to an extent Omron) support this.

The automatic tag reads are only "best effort" and not a guarantee.

ArjanVer, 2023-11-21:
Thanks for your inputs @timyhac and @kyle-github . I will see how to implement the required function.

@kyle-github Yes, unfortunately the PCCC PLCs do not support multiple requests per packet. However, there is something called "block reading", which allows consequent tags to be read at once (or in a more performant way). For example you could read tags like N7:1 to N7:9 in a more performant way (block reading) and not one by one.
Is this functionality available for libplctag? I did some testing but it doesn't seem so, any ideas if this could be implemented easily?

kyle-github, 2024-06-09:
Please reopen if you still have question.

closed 2024-06-09 by kyle-github