libplctag/libplctag #440: Many "INFO" messages output in command prompt after .AddTag(something) command

igal92, 2023-11-19:
Hello,

First i would like to say thank you very much for the libplctag, its very useful and works great with Allen Bradley PLC.
After adding a second tag using the .AddTag method, the command prompt is being filled up with INFO type log endlessly.
attached picture.
Is there any way to disable this?
[image]

kyle-github, 2023-11-19:
It looks like you are turning on debugging at some point.   You probably have a line that looks something like this:

`LibPlcTag.DebugLevel = DebugLevel.Detail;`

somewhere in your code.   Remove that.  Or replace it with:

`LibPlcTag.DebugLevel = DebugLevel.None;`

kyle-github, 2024-04-19:
Please reopen if you still have questions.

closed 2024-04-19 by kyle-github