libplctag/libplctag #446: Use Libplctag Library as Server

aggarwaltapan027, 2023-12-29:
Hi,
Is it possible to use Linbplctag library as a server. If yes, then please tell the changes which we need to make so that it will work as a server. Actually we have a software that works as a Client, that's why we need to make  a server.

Thanks in Advance

kyle-github, 2024-06-09:
It is not currently possible.  There is a very, very rudimentary server in the tests directory called ab_server.  It is just a very minimal simulator that supports a few CIP commands that I use for testing when running in the CI system which does not have any PLCs available.

closed 2024-06-09 by kyle-github

aggarwaltapan027, 2024-06-10:
Hi,
Can you please share the path of that directory.

Thanks a lot for the reply

Regards

Tapan Aggarwal