libplctag/libplctag #448: ErrorBadParam when reading a value from bool dimension

mdByte, 2024-02-01:
We used the following controller:
Vendor: Rockwell Automation/Allen-Bradley
Type: 1756-L81EP ControlLogix 5580 Controller
Revision: 35.011

Prerequisites: an array of bools was created - boolDimension[15];

Let's look at the following cases:
1. Create and try to read boolDimension - was able to successfully read the whole array with all values
2. Create and try to read boolDimension[0] - was able to successfully read value for this dimension at index 0
3. Create and try to read boolDimension[3] - in this case we have ErrorBadParam as for  1,2,3,4,5,6,7,8,9,10,11,12,13,14

we use libplctag.NativeImport version 1.0.33 to work with our controller

Why can't we read the value of an array if the index is greater than zero?

kyle-github, 2024-02-01:
Thanks for using libplctag.   So...  Boolean arrays are weird.   They are packed into DINTs.  So the actual length of a BOOL array is the number of bits divided by 32 rounded up.   In your case that is 1 (one).  To make matters even more strange, the underlying protocol does _NOT_ index based on the array element, but on the packed DINT element.  So the only valid index is 0 (zero).   In order to get the individual BOOL elements, you'll need to use the bit elements if you want to access each one as a separate tag:

boolDimension[0].0 - first BOOL
boolDimension[0].1 - second BOOL
...
boolDimension[0].14 - last BOOL

Note that BOOL arrays are rounded up to the nearest multiple of 32 in length, so you might have up to boolDimension[0].31.

If you want to access just the individual bits in one tag, then use the bit accessor methods.

This is one of the areas where not having the internal tag definition is annoying for users.  I am doing a lot of rethinking about that.   Unfortunately, without getting the full type information, it is pretty much impossible for the library to guess what type (specifically BOOL arrays) the tag is.

I hope this helps!

mdByte, 2024-02-02:
yes, this approach works great, thanks for your help and explanation

kyle-github, 2024-02-02:
Please reopen this issue if you have other problems with this.

closed 2024-02-02 by kyle-github