libplctag/libplctag #450: libplctag never stops after disconnect / reconnect (AB CLX)

joed74, 2024-03-13:
If i interrupt my connection to the plc (disconnect / reconnect) libplctag never returns from plc_tag_shutdown

Attached two logs, first without interruption, second with interruption:

plc_tag_shutdown when connection was not interrupted:
```
2024-03-13 16:07:14.3883  INFO 1 plctag.plc_tag_shutdown:1095 thread(2) tag(0) Starting. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.plc_tag_shutdown:1101 thread(2) tag(0) Closing all tags. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.plc_tag_shutdown:1133 thread(2) tag(0) All tags closed. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.plc_tag_shutdown:1135 thread(2) tag(0) Cleaning up library resources. 
2024-03-13 16:07:14.3883  INFO 1 plctag.ab_teardown:142 thread(2) tag(0) Releasing global AB protocol resources. 
2024-03-13 16:07:14.3883  INFO 1 plctag.ab_teardown:153 thread(2) tag(0) IO thread already stopped. 
2024-03-13 16:07:14.3883  INFO 1 plctag.ab_teardown:156 thread(2) tag(0) Freeing session information. 
2024-03-13 16:07:14.3883  INFO 1 plctag.session_teardown:142 thread(2) tag(0) Starting. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.session_teardown:145 thread(2) tag(0) Waiting for sessions to terminate. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.session_teardown:162 thread(2) tag(0) Sessions all terminated. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.session_teardown:169 thread(2) tag(0) Destroying session mutex. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.mutex_destroy:841 thread(2) tag(0) destroying mutex 6339D904 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.mutex_destroy:854 thread(2) tag(0) Done. 
2024-03-13 16:07:14.3883  INFO 1 plctag.session_teardown:176 thread(2) tag(0) Done. 
2024-03-13 16:07:14.3883  INFO 1 plctag.ab_teardown:162 thread(2) tag(0) Done. 
2024-03-13 16:07:14.3883  INFO 1 plctag.mb_teardown:2427 thread(2) tag(0) Starting. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.mb_teardown:2432 thread(2) tag(0) Waiting for all Modbus PLCs to terminate. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.mb_teardown:2448 thread(2) tag(0) All Modbus PLCs terminated. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.mb_teardown:2452 thread(2) tag(0) Destroying Modbus mutex. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.mutex_destroy:841 thread(2) tag(0) destroying mutex 6339D910 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.mutex_destroy:854 thread(2) tag(0) Done. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.mb_teardown:2456 thread(2) tag(0) Modbus mutex destroyed. 
2024-03-13 16:07:14.3883  INFO 1 plctag.mb_teardown:2458 thread(2) tag(0) Done. 
2024-03-13 16:07:14.3883  INFO 1 plctag.lib_teardown:176 thread(2) tag(0) Tearing down library. 
2024-03-13 16:07:14.3883  INFO 1 plctag.lib_teardown:181 thread(2) tag(0) Signaling tag tickler condition var. 
2024-03-13 16:07:14.3883  INFO 1 plctag.lib_teardown:186 thread(2) tag(0) Tearing down tag tickler thread. 
2024-03-13 16:07:14.3883  INFO 15 plctag.tag_tickler_func:646 thread(1) tag(0) Terminating. 
2024-03-13 16:07:14.3883  INFO 1 plctag.lib_teardown:193 thread(2) tag(0) Tearing down tag tickler condition var. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.cond_destroy:1242 thread(2) tag(0) Starting. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.cond_destroy:1253 thread(2) tag(0) Done. 
2024-03-13 16:07:14.3883  INFO 1 plctag.lib_teardown:199 thread(2) tag(0) Tearing down tag lookup mutex. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.mutex_destroy:841 thread(2) tag(0) destroying mutex 6339D8DC 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.mutex_destroy:854 thread(2) tag(0) Done. 
2024-03-13 16:07:14.3883  INFO 1 plctag.lib_teardown:205 thread(2) tag(0) Destroying tag hashtable. 
2024-03-13 16:07:14.3883  INFO 1 plctag.hashtable_destroy:260 thread(2) tag(0) Starting 
2024-03-13 16:07:14.3883  INFO 1 plctag.hashtable_destroy:272 thread(2) tag(0) Done 
2024-03-13 16:07:14.3883  INFO 1 plctag.lib_teardown:212 thread(2) tag(0) Done. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.mutex_destroy:841 thread(2) tag(0) destroying mutex 6339D8D0 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.mutex_destroy:854 thread(2) tag(0) Done. 
2024-03-13 16:07:14.3883 DEBUG 1 plctag.plc_tag_unregister_logger:1368 thread(2) tag(0) Starting 
```

plc_tag_shutdown when connection was interrupted / disconnected and reconnected:

Now there is a thread(3) ...
```
2024-03-13 15:57:13.5088  INFO 1 plctag.plc_tag_shutdown:1095 thread(2) tag(0) Starting. 
2024-03-13 15:57:13.5088 DEBUG 1 plctag.plc_tag_shutdown:1101 thread(2) tag(0) Closing all tags. 
2024-03-13 15:57:13.5088 DEBUG 1 plctag.plc_tag_shutdown:1133 thread(2) tag(0) All tags closed. 
2024-03-13 15:57:13.5088 DEBUG 1 plctag.plc_tag_shutdown:1135 thread(2) tag(0) Cleaning up library resources. 
2024-03-13 15:57:13.5088  INFO 1 plctag.ab_teardown:142 thread(2) tag(0) Releasing global AB protocol resources. 
2024-03-13 15:57:13.5088  INFO 1 plctag.ab_teardown:153 thread(2) tag(0) IO thread already stopped. 
2024-03-13 15:57:13.5088  INFO 1 plctag.ab_teardown:156 thread(2) tag(0) Freeing session information. 
2024-03-13 15:57:13.5088  INFO 1 plctag.session_teardown:142 thread(2) tag(0) Starting. 
2024-03-13 15:57:13.5088 DEBUG 1 plctag.session_teardown:145 thread(2) tag(0) Waiting for sessions to terminate. 
2024-03-13 15:57:13.5928 DEBUG 17 plctag.session_handler:1162 thread(3) tag(0) in SESSION_IDLE state. 
2024-03-13 15:57:13.6106 DEBUG 15 plctag.tag_tickler_func:636 thread(1) tag(0) Tag tickler thread timed out waiting for something to do. 
2024-03-13 15:57:13.6106  INFO 15 plctag.tag_tickler_func:646 thread(1) tag(0) Terminating. 
2024-03-13 15:57:13.6929 DEBUG 17 plctag.session_handler:1162 thread(3) tag(0) in SESSION_IDLE state. 
2024-03-13 15:57:13.7934 DEBUG 17 plctag.session_handler:1162 thread(3) tag(0) in SESSION_IDLE state. 
2024-03-13 15:57:13.8939 DEBUG 17 plctag.session_handler:1162 thread(3) tag(0) in SESSION_IDLE state. 
2024-03-13 15:57:13.9945 DEBUG 17 plctag.session_handler:1162 thread(3) tag(0) in SESSION_IDLE state. 
2024-03-13 15:57:14.0953 DEBUG 17 plctag.session_handler:1162 thread(3) tag(0) in SESSION_IDLE state. 
2024-03-13 15:57:14.1957 DEBUG 17 plctag.session_handler:1162 thread(3) tag(0) in SESSION_IDLE state. 
[... 130 lines cut ...]
2024-03-13 15:57:22.7459 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:22.8463 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:22.9467 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:23.0472 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:23.1476 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:23.2479 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:23.3486 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:23.4495 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:23.5503 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:23.6512 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:23.7518 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:23.8525 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:23.9526 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:24.0530 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:24.1533 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:24.2536 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:24.3539 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:24.4543 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:24.5552 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:24.6556 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:24.7560 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:24.8568 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:24.9574 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
2024-03-13 15:57:25.0588 DEBUG 17 plctag.session_handler:1270 thread(3) tag(0) in SESSION_WAIT_RECONNECT state. 
```

joed74, 2024-03-20:
```
diff --git a/session.c b/session.c
index 3c36271..60dc163 100644
--- a/session.c
+++ b/session.c
@@ -71,7 +71,7 @@
 #define SOCKET_WAIT_TIMEOUT_MS (20)
 #define SESSION_IDLE_WAIT_TIME (100)
 
-
+volatile int ab_session_terminating = 0;
 
 static ab_session_p session_create_unsafe(const char *host, const char *path, plc_type_t plc_type, int *use_connected_msg, int connection_group_id);
 static int session_init(ab_session_p session);
@@ -141,6 +141,8 @@ void session_teardown()
 {
     pdebug(DEBUG_INFO, "Starting.");
 
+    ab_session_terminating = 1;
+
     if(sessions && session_mutex) {
         pdebug(DEBUG_DETAIL, "Waiting for sessions to terminate.");
 
@@ -173,6 +175,8 @@ void session_teardown()
         session_mutex = NULL;
     }
 
+    ab_session_terminating = 0;
+
     pdebug(DEBUG_INFO, "Done.");
 }
 
@@ -1031,7 +1035,7 @@ THREAD_FUNC(session_handler)
 
     pdebug(DEBUG_INFO, "Starting thread for session %p", session);
 
-    while(!session->terminating) {
+    while(!session->terminating && !ab_session_terminating) {
         /* how long should we wait if nothing wakes us? */
         wait_until_time = time_ms() + SESSION_IDLE_WAIT_TIME;
 
@@ -1321,6 +1325,9 @@ THREAD_FUNC(session_handler)
         purge_aborted_requests_unsafe(session);
     }
 
+    if (ab_session_terminating)
+        session_destroy(session);
+
     THREAD_RETURN(0);
 }
 
```

[session.patch](https://github.com/libplctag/libplctag/files/14667993/session.patch)

I just added an ab_session_terminating flag which will be set in session_teardown and ends all remaining THREAD_FUNCs with additionally destroying the session. Now plctag_shutdown never hangs...

kyle-github, 2024-06-09:
I am just getting back to you, sorry for the delay!

I am not sure why adding an additional flag would make this work.  The existing `session->terminating` flag should be sufficient.   

I have not been able to reproduce this.   Are the steps here:

1. let libplctag connect to a PLC.
2. interrupt the connection (by turning off the PLC? by pulling the Ethernet cable?)
3. terminate the session by closing the last tag.

Is that correct?

kyle-github, 2024-08-27:
I found some places in the main API where the tag was not getting released and I think that contributed to this as the destructor for the session object was not getting called.  I think this is fixed with version 2.6.2.  Please check your use case and reopen this if it is still not working.

Thanks again for the bug report!

commit c55bc587 (2024-09-03): Prerelease merge into release (#491)

* Make file number optional if input or output file.  Fix #481.

* Added fix for omitted data file number on I or O files. Added support for a numeric subelement.  Fix #479. (#483)

* Fix #486. (#487).   String size change problem.   Fixed with #488.

* Added atomic flag and checks to aid in making all sessions terminate when the library is shutting down. Fix #450.

* plc_tag_set_string() feature additions (#488).   Setting string can change tag data buffer size.  And added flag, allow_field_resize, to suppress change.

* Added msp response size check befo [... cut]
closed 2024-08-27 by kyle-github