libplctag/libplctag #452: Multiple request are not packed into single request for Omron NX/NJ PLC

ahmad9918, 2024-04-02:
Hello Kyle,

I am trying to use libplctag library to check if multiple request are getting packed into single request in order to increase performance when reading more number of tags. For Omron NX/NJ PLC, I found out that library is not packing multiple request into single request even if Large Forward Open is success. But when I try to forcefully set the value to true(1) for tag->allow_packing field, then it goes for MSR packing.

Can you please suggest why library is not going for multiple request packing even if we have received success response for large forward open request with connection parameter as 1996 bytes. Please find Wireshark screenshot for reference :

[image]

kyle-github, 2024-04-19:
Hmm, it should be I think.   Let me check the code...

```c
     tag->allow_packing = attr_get_int(attribs, "allow_packing", 0);
```

Looks like it is defaulting to not allow packing.   I must have done this for a reason.  I think the problem is that Omron does not allow partial results when using the Rockwell standard CIP commands.  So if the results exceed the size of the allowed result packet, then the request will fail.

kyle-github, 2024-06-09:
This is related to issue #454 and the outstanding PR by @ptsOSL.  I hope to land more Omron support in the next week or so.  Please reopen if that does not solve this for you.