libplctag/libplctag #456: Standalone website to host documentation

timyhac, 2024-04-16:
Is there value in moving content from the Wiki to a standalone website?

I was fiddling around with Github's pages feature and a Static Site Generator called Docusaurus and used libplctag documentation as a test case - I think it gives the library that extra bit of polish.

Published site on Github Pages: https://timyhac.github.io/libplctag-docs/docs/intro
The source content itself is still [mostly markdown](https://docusaurus.io/docs/markdown-features): https://github.com/timyhac/libplctag-docs/tree/main/docs

Pros:
* Polish
* Easier to read and navigate?
* Better SEO

Cons:
* Extra complexity - Github's wiki system is already pretty great. Another great option is just store flat markdown files in a [/docs/ folder](https://github.com/libplctag/libplctag.NET/tree/master/docs).
* Github has search built-in. Docusaurus requires setup with Algolia.

While porting the content over I made a few minor changes to the structure of the documents - grouping information about API usage together - this change I think would be useful regardless of whether we go ahead with this or not.

I thought I'd share this in case people have other opinions but I'm leaning towards it being overkill.

kyle-github, 2024-04-19:
Didn't we talk about this a couple years ago?   I do like the idea somewhat.   I am actually leaning toward a docs directory within the repo though.  The reason for that is that I can version the docs with the code so that, for example, additions of functions will be documented along with the specific code that includes them.   I will look into Docusaurus.   I had just stuck with MD because it is fairly universal, but it is not that great when someone checks out the repo.

timyhac, 2024-04-19:
100% Kyle - the value of keeping docs and code together is really undervalued and you'd have to do some gymnastics to make it work with an external documentation site.
I'll keep the repo up for a little while but take the site down as I don't want it to be mistaken for the real documentation.

closed 2024-04-19 by timyhac

kyle-github, 2024-04-20:
I like how you formatted your example.   Having more than one level is nice.   Maybe I can leverage that idea if I make a docs directory by having some sub folders?

timyhac, 2024-04-21:
Astro.build is also pretty nice - https://starlight.astro.build/getting-started/