libplctag/libplctag #457: Write is not working for SINT and USINT (1 byte) datatype for Omron NX/NJ PLC

ahmad9918, 2024-04-19:
Hello Kyle,

I am using libplctag library in order to read and write for SINT and USINT datatype for Omron NX/NJ plc . I have observed read is working as expected but while writing value for SINT and USINT tag, an extra byte is written into the PLC which results into error being returned in the response. I found an extra byte is getting written while analyzing wireshark log. In order to verify if an extra byte is causing problem I tried to modify the packet structure by removing the extra byte also adjusting the length of the buffer and sent using socket sample application. It resulted into proper response from the PLC with value getting written into the PLC.

Can you please share your thoughts on this issue and provide the fix for these datatypes.
Thanks

kyle-github, 2024-05-26:
I think I may have an idea here.   I found these statements in another manual:

- Omron’s NJ structure types use the natural alignment of their members, up to 64-bit, then pad
the structure to the largest alignment of it the members. Which then becomes the structure’s
natural alignment when nesting in other structures. Named booleans in NJ structures occupy
and align to a complete 16-bit word of their own.
- Omron’s CJ structure types use only 16-bit alignment, **even for bytes and strings**. Consecutive
named booleans are packed into 16-bit words together, and in this library, must be hosted in a
16-bit word.

Right now the library does not distinguish between CJ and NJ PLC types.   That seems to be a problem due to alignment.

kyle-github, 2024-06-08:
I think that this is going to be fixed by @ptsOSL's PR.  Once that is merged, we will need to check.

ptsOSL, 2024-06-13:
Yes, I have tested this with the prerelease version and reading and writing SINT and USINT is working for me.

kyle-github, 2024-06-14:
Thanks!  Getting about time to release the ~~Kraken~~version.

kyle-github, 2024-06-14:
Calling this closed with release v2.6.0.  Please reopen if there are still problems.

closed 2024-06-14 by kyle-github