libplctag/libplctag #458: MSR is not working for modbus_tcp protocol using libplctag library

salma4791, 2024-04-23:
I am using libplctag to communicate with Quantum PLC CPU 140 . I am able to establish communication and read the tags (Absolute Addressing). For multiple tags, i have observed that the requests are not unified.. So the requests are going one after the other and not as a **Multi Service Request**.  So when i try to configure 400 or 500 tags, it is taking lot time to read. Basically the performance is not very good with the single request for so many tags. Any way that libplctag can support MSR for modbus_tcp protocol?

kyle-github, 2024-06-01:
Hi, sorry for the delay. I missed this issue.

I have not found any documentation about a multiple service request Modbus command.   Do you mean queuing multiple requests together and sending them at once via TCP?  If so, I have experimented with that.  The Modbus protocol specification allows for up to 16 requests "in flight" in that way.   However, only some devices support that feature.

salma4791, 2024-06-06:
Yes I meant queuing multiple requests and sending them at once similar to the "allow_packing" in Allen brandly.. The Plc which i am using is Quantum 140 Modbus PLC (Schneider) . This hardware supports multiple requests "in flight" but using libplctag, library , I see only single request is being sent at a time.

kyle-github, 2024-06-09:
There is an attribute you can set in the tag string, "max_requests_in_flight", that controls this.   As per the Modbus specification the value can get 1-16 (inclusive).  The default is 1.  

This controls the number of requests that are sent to the device without waiting for a response.   

An example tag string:
```
protocol=modbus-tcp&gateway=127.0.0.1&path=0&elem_count=2&name=hr10&max_requests_in_flight=6
```

Note that this feature is not heavily tested, so please let me know if it does not work as expected!

closed 2024-06-09 by kyle-github