libplctag/libplctag #459: MinGW timeapi.h issue

JKooAllied, 2024-05-02:
Hello,

I followed Wiki. After mingw32-make, I am getting this issue: 
libplctag\src\platform\windows\platform.c:48:21: fatal error: timeapi.h: No such file or directory
 #include <timeapi.h>

kyle-github, 2024-05-02:
Hmm.  It looks like it is missing in some versions of MinGW.  I found this notice on the main MinGW SourceForge site:

https://sourceforge.net/p/mingw-w64/mailman/message/37409800/

That is from 2022.  It looks like timeapi.h should be included with MinGW.   It is one of the header files provided with Visual C/C++ as well.   

When was your version of MinGW created?

kyle-github, 2024-06-09:
Please reopen if you still have a problem.  This question is probably better answered on the [forum](https://groups.google.com/g/libplctag).  There are some people on it who use MinGW.

closed 2024-06-09 by kyle-github