libplctag/libplctag #462: Erroneous warning when creating Omron session

ptsOSL, 2024-06-06:
Hi Kyle, just a minor issue here. 
I am seeing this warning : 

> WARN match_numeric_segment:210 Numeric segment in path at position 0 is out of bounds!

This is coming from cip.c, see the code snippet below:

```c
    /* was the numeric segment valid? */
    if(val < 0 || val > 0x0F) {
        pdebug(DEBUG_WARN, "Numeric segment in path at position %d is out of bounds!", (int)(ssize_t)(*path_index));
        return PLCTAG_ERR_OUT_OF_BOUNDS;
    }
``` 

The path which is used for Omron, value 18, is larger than this if statement expects and so it returns this warning. I am pretty sure that a value of 18 is fine, and that this is just outdated for Omron. There is no actual effect on the code, except to print this warning, the return value is just used to print another debug message.

If you think this sounds right, I will change val > 0x0F to val > 0x12, in the Omron updates branch

ptsOSL, 2024-06-06:
Now that I have tested my solution, I have realised that I have misunderstood something here as I am now getting errors.

ptsOSL, 2024-06-06:
Ok having looked at it some more, I think its working fine, its just that I am giving it an IP address instead of a numeric segment. So when it tries to find a valid numeric segment it returns this warning. It seems that using a numeric segment is optional, would it be better to print an INFO debug statement rather than a WARN?

kyle-github, 2024-06-07:
Here are some notes on how better to handle this.

There are multiple types of segment.   Segments are port and link pairs, roughly.

P L - each can be more than one byte

Types:

**000 - port segment**

000 E PPPP [ EEEEEEEE ] [ IIIIIIIIIIIIIIII ]
 
E = extended link flag.  If set, then the EEEEEEEE byte is there and contains the length of the link part in bytes.   

PPPP - 1-14 for the port number.  1 is always the backplane.  If this is 15, then the optional IIIIIIIIIIIIIIII 16-bit word contains the port identifier.

Format is e pppp where e is the extended link flag and pppp is the port identifier.   If the extended link flag is set, then there is another byte following this one that indicates the number of bytes the link part.

The port identifier is 0-14.  If it is 15, then a 16-bit port identifer word appears after the optional extended link length byte

**001 - logical segment**

001 TTT FF [ EEEEEEEE ] <logical val, 1, 2 or 4 bytes>

TTT - logical segment type

000 - class ID
001 - instance ID
010 - member ID
011 - Connection point
100 - attribute ID
111 - extended logical type

FF - size of the logical value

00 - 1 byte
01 - 2 bytes
10 - 4 bytes

EEEEEEEE - extended logical type if the TTT field is 111

0 - Reserved
1 - Array Index
2 - Indirect Array Index
3 - Bit Index
4 - Indirect Bit Index
5 - Structure Member Number
6 - Structure Member Handle

A Bit Index Extended Logical Segment is used to specify the 0-based bit number into the preceding item in the EPath (there is no restriction on type of the preceding item).

A Structure Member Number Extended Logical Segment is used to specify the 1-based member number into the preceding item in the EPath which must be a structure. The member number specifies the structure member, where the structure members are numbered starting with 1.

A Structure Member Handle Extended Logical Segment is used to specify the member handle into the preceding item in the EPath which must be a structure. The handle is a value that uniquely identifies a member and is provided by the target device when reporting constructed data type information using the Formal Encoding with Handles specified in section C-6.2.1.

**010 - Network segment**

Not needed?

**011 - Symbolic Segment**

011 sssss [ fffsssss ]

sssss - 1-31 size of the symbolic segment in bytes.  If it is 0 the optional format/size byte follows.

fffsssss - fff is the format and sssss is the number of elements.

**100 10001** - Extended ANSI Symbolic Segment, 0x91

This is the one used for all tag names usually.  It is followed by a byte containing the number of 8-bit characters in the segment data.

kyle-github, 2024-06-14:
The real problem is that each segment in a path is a pair of port/node.   The port part is relatively limited to smaller numbers (<16 IIRC) for standard port numbers.  The node part, on the other hand is not.   This is part of the issue with a user who has a very large ControlLogix chassis with more than 16 slots.

This parsing needs to be rewritten to be a lot smarter.  I am going to call this a bug.

kyle-github, 2024-06-14:
This should solve #443.

kyle-github, 2024-08-27:
I've been trying to figure out how to do this.  The problem is that we have just a number.  You cannot tell whether that number is supposed to be a port segment or something else.  I.e. "18".  What does that mean.  Is it a long port segment?  Or is it an extended link segment?   That's what you use in front of an IP address, for instance.  It looks like I'll need to come up with some alternate notation.

kyle-github, 2024-08-27:
Hmm, I think I am overthinking this.  I just need to parse the path in pairs.  port/link is one such pair.  The second part does not need to be less than 0x0f.

kyle-github, 2025-03-04:
Changing this to an enhancement.   This needs to happen soon.

closed 2025-03-04 by kyle-github