libplctag/libplctag #464: Read/writing OmronNJ/NX TIME and DATE_AND_TIME data

ptsOSL, 2024-06-10:
Hi Kyle, hopefully this is the last change needed for Omron NJ/NX support.

Omron NJ/NX claims to support the standard CIP TIME datatype with code 0xDB, but I have not yet managed to read a variable with this datatype. Instead, when reading TIME variables, Omron defaults to using their "Vendor Specific" TIME_NSEC datatype instead. I have sent an information request to their tech support for more info about this.

Anyway, I was wondering if it would be possible to add support for two Omron NJ/NX "Vendor Specific" datatypes. These being TIME_NSEC (0x09) and DATE_AND_TIME_NSEC (0x0A). Is adding "Vendor Specific" datatypes something that you have done for other PLCs and how difficult do you think it would be to add them to the library? 

They are both 8 byte datatypes and I believe (but am for from certain) that they contain a ULINT which counts nanoseconds since 1970-01-01-00:00:00. I expect that they would just be read into a libplctag tag and the data could then be accessed using the plc_tag_get_uint64() function.

Cheers,
Phil

kyle-github, 2024-06-10:
I am not sure what you mean by supporting these.  The library, currently, does nothing at all with the type that comes from the PLC.  It just treats all tags as big bags of bytes.   That said, I am working on an extension to the library to return the native PLC type data on request.   That should help with translating the data.

ptsOSL, 2024-06-10:
Sorry, I should have been more specific. If you attempt to read one of these datatypes, as they are not standard CIP datatypes, a warning is initially raised in eip_cip.c and then PLC_ERR_UNSUPPORTED is returned. The debug messages received are:

> WARN check_read_status_connected:1496 Unsupported data type returned, type byte=10

> WARN check_read_status_connected:1579 Error received!

> WARN plc_tag_create_ex:1047 Error PLCTAG_ERR_UNSUPPORTED while trying to create tag!

From looking at the code, it seems that this is because the code associated with the datatype is not in the list of #defines in defs.h. This is checked on line 1463 of eip_cip.c. This is the only support I was really taking about. But I wasn't sure if it was enough to simply add the datatypes to the defs.h file, making it clear that they are specific to Omron. I will look into it some more.

kyle-github, 2024-06-10:
Ah... I see now.   Hmm, I forgot I was doing that.   Here's the code (starting around line 1417 in my copy):

```c
            if ((*data) >= AB_CIP_DATA_BIT && (*data) <= AB_CIP_DATA_STRINGI) {
                /* copy the type info for later. */
                if (tag->encoded_type_info_size == 0) {
                    tag->encoded_type_info_size = 2;
                    mem_copy(tag->encoded_type_info, data, tag->encoded_type_info_size);
                }

                /* skip the type byte and zero length byte */
                data += 2;
            } else if ((*data) == AB_CIP_DATA_ABREV_STRUCT || (*data) == AB_CIP_DATA_ABREV_ARRAY ||
                       (*data) == AB_CIP_DATA_FULL_STRUCT || (*data) == AB_CIP_DATA_FULL_ARRAY) {
```

This is a bit problematic.   I don't want to have the whole range of type IDs allowed, but maybe a look up array?  It would only be 256 entries.  Let me play with that idea.  Then we can just fix these up as we go and make it all work.

kyle-github, 2024-06-11:
This is in the prerelease branch.  I took the idea of a lookup table.  It is in cip.c.   I found a few more things to add to it.   Not sure of the sizes though.  Here's the table:

```c
struct cip_type_lookup_entry_t {
    int is_found;
    int type_data_length;
    int instance_data_length;
};

static struct cip_type_lookup_entry_t cip_type_lookup[] = {
    /* 0x00 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x01 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x02 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x03 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x04 */ { PLCTAG_STATUS_OK,    2, 2 },   /* UINT_BCD: OMRON-specific */
    /* 0x05 */ { PLCTAG_ERR_NO_MATCH, 2, 4 },   /* UDINT_BCD: OMRON-specific */
    /* 0x06 */ { PLCTAG_ERR_NO_MATCH, 2, 8 },   /* ULINT_BCD: OMRON-specific */
    /* 0x07 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },   /* ???? ENUM: OMRON-specific */
    /* 0x08 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },   /* ???? DATE_NSEC: OMRON-specific */
    /* 0x09 */ { PLCTAG_STATUS_OK,    2, 8 },   /* TIME_NSEC: OMRON-specific, Time in nanoseconds */
    /* 0x0a */ { PLCTAG_STATUS_OK,    2, 8 },   /* DATE_AND_TIME_NSEC: OMRON-specific, Date/Time in nanoseconds*/
    /* 0x0b */ { PLCTAG_ERR_NO_MATCH, 0, 0 },   /* ???? TIME_OF_DAY_NSEC: OMRON-specific */
    /* 0x0c */ { PLCTAG_ERR_NO_MATCH, 0, 0 },   /* ???? UNION: Omron-specific */
    /* 0x0d */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x0e */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x0f */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x10 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x11 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x12 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x13 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x14 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x15 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x16 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x17 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x18 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x19 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x1a */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x1b */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x1c */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x1d */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x1e */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x1f */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x20 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x21 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x22 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x23 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x24 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x25 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x26 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x27 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x28 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x29 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x2a */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x2b */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x2c */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x2d */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x2e */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x2f */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x30 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x31 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x32 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x33 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x34 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x35 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x36 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x37 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x38 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x39 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x3a */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0x3b */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
[... 176 lines cut ...]
    /* 0xec */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xed */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xee */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xef */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xf0 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xf1 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xf2 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xf3 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xf4 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xf5 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xf6 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xf7 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xf8 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xf9 */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xfa */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xfb */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xfc */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xfd */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xfe */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
    /* 0xff */ { PLCTAG_ERR_NO_MATCH, 0, 0 },
};

```

Please add entries that you know about!

ptsOSL, 2024-06-11:
Thanks for this Kyle. I have been testing reading the Omron TIME and DATE_AND_TIME variables on the prerelease branch. Its working great so far. I dont plan on using the other Omron data types in my work, but ill do some test read/writes of them anyway and see if I can fill in those entries in the lookup table.

kyle-github, 2024-06-11:
Can you please see if you can read/write SINT and USINT and BOOL tags too?

Also if you could make a BOOL array and capture the debug (level 4) output when you read it, I would appreciate it.  I think that Omron transfers these in a different way than Rockwell.  Of course.

ptsOSL, 2024-06-11:
I have tested the remaining two Omron time variables, they are both 8 bytes with 2 byte type_data_length. Updating the structure appropriately and read/writing these works fine. As a note for others reading this, all four datatypes (at least on Omron NJ), return the number of nanoseconds between 1970-01-01 and the value stored in the PLC variable. I believe the time is based off the CPUs configured timezone rather than UTC, but I have not tested changing the timezone to test. The data type returned is actually a signed 64 bit integer, not unsigned.

Although I see the BCD types in the manual for my PLC, I do not seem to be able to create variables with this type, so cant test read/writing these. But im pretty sure they would be 2,4 and 8 bytes respectively, maybe we should allow these anyway?

Read/writing SINT and USINT both work fine.

Read/writing single booleans works fine.

This is the result from reading an array of 20 booleans, with values: [0,1,0,1,0,1,0,1,0,1,0,1,0,1,0,1,0,1,0,1]

```
2024-06-11 17:15:30.967 thread(3) tag(0) DETAIL socket_read:1814 Socket can read data.
2024-06-11 17:15:30.967 thread(3) tag(0) DETAIL socket_read:1866 Done: result 24.
2024-06-11 17:15:30.967 thread(3) tag(0) DETAIL socket_read:1754 Starting.
2024-06-11 17:15:30.967 thread(3) tag(0) DETAIL socket_read:1866 Done: result 32.
2024-06-11 17:15:30.967 thread(3) tag(0) INFO recv_eip_response:2023 request received all needed data (56 bytes of 56).
2024-06-11 17:15:30.967 thread(3) tag(0) INFO recv_eip_response:2025 00000 70 00 20 00 86 01 5b 4e 00 00 00 00 00 00 00 00
2024-06-11 17:15:30.967 thread(3) tag(0) INFO recv_eip_response:2025 00016 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00
2024-06-11 17:15:30.967 thread(3) tag(0) INFO recv_eip_response:2025 00032 a1 00 04 00 72 47 1a 7d b1 00 0c 00 02 00 cc 00
2024-06-11 17:15:30.967 thread(3) tag(0) INFO recv_eip_response:2025 00048 00 00 c1 00 55 55 05 00
2024-06-11 17:15:30.967 thread(3) tag(0) INFO recv_eip_response:2032 Done.
2024-06-11 17:15:30.967 thread(3) tag(11) INFO unpack_response:1558 Starting.
2024-06-11 17:15:30.967 thread(3) tag(11) INFO unpack_response:1567 Got single response packet. Copying 56 bytes unchanged.
2024-06-11 17:15:30.967 thread(3) tag(11) INFO unpack_response:1656 Unpacked packet:
2024-06-11 17:15:30.967 thread(3) tag(11) INFO unpack_response:1657 00000 70 00 20 00 86 01 5b 4e 00 00 00 00 00 00 00 00
2024-06-11 17:15:30.967 thread(3) tag(11) INFO unpack_response:1657 00016 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00
2024-06-11 17:15:30.967 thread(3) tag(11) INFO unpack_response:1657 00032 a1 00 04 00 72 47 1a 7d b1 00 0c 00 02 00 cc 00
2024-06-11 17:15:30.967 thread(3) tag(11) INFO unpack_response:1657 00048 00 00 c1 00 55 55 05 00
2024-06-11 17:15:30.967 thread(3) tag(11) DETAIL unpack_response:1666 Done.
``` 
I think this is working as expected. It is storing each bit in a single bit, then padding to a byte, and then padding the whole array to multiples of two bytes.

ptsOSL, 2024-06-11:
I still have the Omron Unions and Enums to test, these are both struct like and I they can have variable sizes.

Here is the wireshark capture from reading the boolArray as well:
[image]

kyle-github, 2024-06-11:
Thanks for these tests!

ptsOSL, 2024-06-12:
I have tested the enum type, this has a 2 byte type_data_length and a 4 bytes instance_data_length. It is essentially just a 4 byte integer which points to an enumeration string within the plc. 

The union I have been unable to get working. I can read members from a union, but I am unable to read the union itself. I get a vendor specific error:

WARN check_read_status_connected:1416 CIP read failed with status: 0x1f Unknown error code.

This is a "vendor specific" error, with status 8007.

In the Omron manual, this is described as "8007 An inaccessible variable was specified". I dont know why this is happening, and as Unions seem fairly unimportant, I dont think it is worth fixing unless someone comes along specifically requesting to access Unions. I would just leave this code as invalid within the lookup table. What are your thoughts on this?

This is how I think the Omron section should look:
```
    /* 0x04 */ { PLCTAG_STATUS_OK,    2, 2 },   /* UINT_BCD: OMRON-specific */
    /* 0x05 */ { PLCTAG_STATUS_OK,    2, 4 },   /* UDINT_BCD: OMRON-specific */
    /* 0x06 */ { PLCTAG_STATUS_OK,    2, 8 },   /* ULINT_BCD: OMRON-specific */
    /* 0x07 */ { PLCTAG_STATUS_OK,    2, 4 },   /* ENUM: OMRON-specific */
    /* 0x08 */ { PLCTAG_STATUS_OK,    2, 8 },   /* DATE_NSEC: OMRON-specific */
    /* 0x09 */ { PLCTAG_STATUS_OK,    2, 8 },   /* TIME_NSEC: OMRON-specific, Time in nanoseconds */
    /* 0x0a */ { PLCTAG_STATUS_OK,    2, 8 },   /* DATE_AND_TIME_NSEC: OMRON-specific, Date/Time in nanoseconds*/
    /* 0x0b */ { PLCTAG_STATUS_OK,    2, 8 },   /* TIME_OF_DAY_NSEC: OMRON-specific */
    /* 0x0c */ { PLCTAG_ERR_NO_MATCH, 0, 0 },   /* Currently unavailable. UNION: Omron-specific */
``` 
Would you like me to commit this, or perhaps it would be easier if you did it?

ptsOSL, 2024-06-12:
Sounds great and yeah its much cleaner now

kyle-github, 2024-06-12:
I updated the prerelease branch.

kyle-github, 2024-06-17:
This was released with v2.6.0.

closed 2024-06-17 by kyle-github